carlosperate/ardublockly #114: external type for block

SebCanet, 2016-08-19:
If you have a look to ma Blockly@rduino repo, you will see I tried to make easy to class/update files/scripts from you, me, and Blockly.
The idea is to make very easy updating !
And that's why I propose a simple way to change the way you type blockly blocks.
In m index.html, after all blocks call js, I put the 'blokcs_typing.js' which store every kind o type you add, just with this

```
Blockly.Blocks.logic_compare.getBlockType = function() {
	return Blockly.Types.BOOLEAN;
};
```
This way it's more modular. Hope it could help...

carlosperate, 2016-08-24:
Thanks again for the suggestion @SebCanet !
As with the other one, I will have a proper look once I'm back.

Any reason to have these as a separate file instead of the block js files? Is it for simpler maintenance with upstream Blockly?
How do you deal with blocks that might need type recognition on runtime? Things like a block type depending on child block types and then the child depending on user input (like the number block being able to be a integer or float)?

SebCanet, 2016-08-25:
it is for simpler maintenance with upstream Blockly, this way even newbies could update, and try to make it simply without documentation.
For math_number:

Blockly.Blocks.math_number.getBlockType = function() {
    var numString = this.getFieldValue('NUM');
    return Blockly.Types.identifyNumber(numString);
};

And control fo:

	Blockly.Blocks.controls_for.getVarType = function() {
		return this.inputList[1].connection.targetBlock().getBlockType();
	};
Hope it answers...