carlosperate/ardublockly #116: volatile number type for interruption

SebCanet, 2016-08-19:
Missing in types.js, only for interruption
[https://www.arduino.cc/en/Reference/AttachInterrupt](https://www.arduino.cc/en/Reference/AttachInterrupt)

```
/** Volatile integer number. */
Blockly.Types.VOLATIL_NUMBER = new Blockly.Type({
  typeId: 'Volatil Number',
  typeMsgName: 'ARD_TYPE_VOLATILE',
  compatibleTypes: []    // Circular dependencies, add after all declarations
});
```

In generaor arduino.js
```
    case Blockly.Types.VOLATIL_NUMBER.typeId:
      return 'volatile int';
```

carlosperate, 2016-08-24:
Thanks for the report @SebCanet.
Boy that's an interesting one! One of the concepts I would love to maintain in Ardublockly is to reduce or hide, or automatically implement, as much as possible the features that might require careful explanation.

I haven't implemented interrupts yet, so I didn't come across this requirement so far. How do you use volatile in your implementation? as a selectable variable type and then instruct users that they should use it on any variable accessed by interrupts?

SebCanet, 2016-08-25:
Because it's a real specific functionnality, it's only a selectable variable.
And I'm totally agree with you about "automatically implement, as much as possible the features that might require careful explanation". That will be a big part of our job, but you already did a lot with typing variable.