carlosperate/ardublockly #117: were the instructions ever created for this point

tcampbellsoup, 2016-08-24:


carlosperate, 2016-08-24:
The [Configure Ardublockly wiki page](https://github.com/carlosperate/ardublockly/wiki/Configure-Ardublockly) hasn't been done yet, but the wiki is open, so you can always give it a go if you fancy contributing :)

I must admit I've never heard about the Machido boards, do you have a link? are they Arduino compatible? Ardublockly uses the Arduino IDE for compiling and uploading, so if it's supported by the Arduino IDE it would be quite straightforward to add support in Ardublockly, all we need is the general board info and the IDE CLI compiler flags.

tcampbellsoup, 2016-08-25:
Machido boards are Arduino compatible, you can use the Arduino IDE. Don't really have a site for the Machido boards, just ordered a bulk supply from the company for some projects. I will take a look at your wiki page tomorrow and get back to you if I am still experiencing some difficulties.

carlosperate, 2016-08-25:
I couldn't find any specific information on their website. Do you have a
link for the stuff you bought?