carlosperate/ardublockly #118: uncompressed js question

huangganggui, 2016-08-25:
Hi,
I come across a question when I modify the file **ardublocky > index.html.**
I changed the script src to use uncompressed version and comment in uncompressed js.  
when I run the start.py, the problems occur. I chose the Motor category, and console log "Uncaught TypeError: Blockly.FieldIntance is not a constructor      stepper.js". 
I searched the git log and find that the stepper was changed, I want to figure out why the compressed js works well but uncompressed js not. 
Thanks a lot.

carlosperate, 2016-08-25:
See if any of the developer FAQs help in your situation: https://github.com/carlosperate/ardublockly/wiki/FAQs

How did you download Ardublockly? was it from a release, the build server, or cloned from the repository? Could you send me a copy or a link to your index.html? You need to make sure you comment-out the 3 compressed sources, and uncomment-in all the uncompressed ones.

huangganggui, 2016-08-30:
My bad.
I missed the step git submodule.
Thanks a lot for your guidance.

closed 2016-08-30 by huangganggui