carlosperate/ardublockly #121: Can't Find Port

ereedsanchez, 2016-10-06:
I am using ArduBlockly on Mac, and the software can not see the a com port. 

I would also like to start a separate conversation with you regarding using Ardublockly in a classroom environment.

carlosperate, 2016-10-06:
Does everything else work as expected? Are you running the application or the command line? which way have you downloaded Ardublockly? Does it show any port at all?

ereedsanchez, 2016-10-08:
No it says no ports avaliable. I downloaded the version for Mac. I DID NOT
use the git to download the whole repo, and run with command line.

Arduino IDE does work by itself, shows ports, and I can download sketches
to the board.

Thanks,

Edwin

carlosperate, 2016-10-09:
Could you post the contents of `Ardublockly.app\Contents\ardublockly.log` to see if there were any errors there?

rp10007, 2016-10-21:
I'm having the same problem. The log shows that serial port = none, and if I modify the server settings file to make the port as detected in the arduino IDE, there's still an error, prefs still says no serial ports available, and then the ini file gets rewritten with serial port = none again.

Meanwhile, some possibly useful log information:

2016-10-21T02:42:09.939Z - info: [ArdublocklyElec] Starting Ardublockly version: 0.1.2
2016-10-21T02:42:09.941Z - info: [ArdublocklyElec] Ardublockly root dir: /Users/xx/Downloads/ardublockly/Ardublockly.app/Contents
2016-10-21T02:42:09.941Z - info: [ArdublocklyElec] OS detected: darwin
2016-10-21T02:42:10.040Z - info: [ServerMgr] Command: /Users/xx/Downloads/ardublockly/Ardublockly.app/Contents/arduexec/server/start --findprojectroot --nobrowser
2016-10-21T02:42:10.343Z - info: [ArdublocklySrv] Running Python 3.5.2 (64 bit) on Darwin-15.6.0-x86_64-i386-64bit

======= Parsing Command line arguments =======

Parsed "--findprojectroot" flag. The ardublockly project root will be set as the server root directory.
Parsed "--nobrowser" flag. No browser will be opened.

======= Resolving server and project paths =======

Ardublockly root directory: /Users/xx/Downloads/ardublockly/Ardublockly.app/Contents
Selected server root: /Users/xx/Downloads/ardublockly/Ardublockly.app/Contents

======= Loading Settings =======

Settings loaded from:
	 /Users/xx/Downloads/ardublockly/Ardublockly.app/Contents/ServerCompilerSettings.ini

Settings file Serial Port is not currently available:
	None
Default Serial Port set:
	None

Final settings loaded:
	Compiler directory: /Users/xx/Downloads/Arduino.app/Contents/MacOS/Arduino
	Arduino Board Key: Uno
	Arduino Board Value: arduino:avr:uno
	Serial Port Value: None
	Sketch Name: ArdublocklySketch
	Sketch Directory: /Users/xx/Documents/Arduino
	Load IDE option: upload

Settings file saved to:
	/Users/xx/Downloads/ardublockly/Ardublockly.app/Contents/ServerCompilerSettings.ini

2016-10-21T02:42:10.344Z - info: [ArdublocklySrv] 

======= Starting Server =======

Setting HTTP Server Document Root to: 
	/Users/xx/Downloads/ardublockly/Ardublockly.app/Contents

2016-10-21T02:42:10.345Z - error: [ArdublocklySrv] Failed to execute script start
Traceback (most recent call last):
  File "start.py", line 176, in <module>
    main()
  File "start.py", line 172, in main
    ardublocklyserver.server.start_server(server_root)
  File "ardublocklyserver/server.py", line 29, in start_server
    server = BaseHTTPServer.HTTPServer(server_address, BlocklyRequestHandler)
  File "socketserver.py", line 440, in __init__
  File "http/server.py", line 138, in server_bind
  File "socketserver.py", line 454, in server_bind
OSError: [Errno 48] Address already in use

2016-10-21T02:42:10.393Z - info: [ArdublocklySrv] Process exited with code 255

rp10007, 2016-11-28:
I was playing with this some more and it appears the problem is definitely somewhere on the ardublockly side. If I click into the underlying Arduino IDE and upload the sketch from there, it works fine.

carlosperate, 2016-11-28:
Hi @rp10007,
The previous error log that you've included failed because the server was already running in the background. Could you close everything, open ardublockly, and then capture the log when this issue appears? That would help me a lot when looking into this.

rp10007, 2016-11-28:
Thanks! Here you go:

2016-11-28T16:40:56.846Z - info: [ArdublocklyElec] Starting Ardublockly version: 0.1.2
2016-11-28T16:40:56.847Z - info: [ArdublocklyElec] Ardublockly root dir: /Applications/Ardublockly.app/Contents
2016-11-28T16:40:56.848Z - info: [ArdublocklyElec] OS detected: darwin
2016-11-28T16:40:57.105Z - info: [ServerMgr] Command: /Applications/Ardublockly.app/Contents/arduexec/server/start --findprojectroot --nobrowser
2016-11-28T16:40:58.925Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-11-28T16:40:59.402Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-11-28T16:41:01.299Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2016-11-28T16:41:01.631Z - info: [ArdublocklySrv] Running Python 3.5.2 (64 bit) on Darwin-15.6.0-x86_64-i386-64bit

======= Parsing Command line arguments =======

Parsed "--findprojectroot" flag. The ardublockly project root will be set as the server root directory.
Parsed "--nobrowser" flag. No browser will be opened.

======= Resolving server and project paths =======

Ardublockly root directory: /Applications/Ardublockly.app/Contents
Selected server root: /Applications/Ardublockly.app/Contents

======= Loading Settings =======

Settings file corrupted or not found in:
	 /Applications/Ardublockly.app/Contents/ServerCompilerSettings.ini

Settings will be set to the default values.

Final settings loaded:
	Compiler directory: None
	Arduino Board Key: Atmel atmega168pb Xplained mini
	Arduino Board Value: atmel:avr:atmega168pb_xplained_mini
	Serial Port Value: None
	Sketch Name: ArdublocklySketch
	Sketch Directory: /Applications/Ardublockly.app/Contents
	Load IDE option: open

Settings file saved to:
	/Applications/Ardublockly.app/Contents/ServerCompilerSettings.ini

2016-11-28T16:46:10.226Z - info: [ArdublocklySrv] 

======= Starting Server =======

Setting HTTP Server Document Root to: 
	/Applications/Ardublockly.app/Contents

Launching the HTTP service...
The "something" = ['else'] parameter is not recognised!

Compiler file in OS X located within the app bundle.

Compiler directory set to:
	/Applications/Arduino1610.app/Contents/MacOS/Arduino
Settings file saved to:
	/Applications/Ardublockly.app/Contents/ServerCompilerSettings.ini

2016-11-28T16:46:15.909Z - info: [ArdublocklySrv] The "something" = ['else'] parameter is not recognised!

Provided Serial Port is not valid: !!!
	no_ports
Default Serial Port set:
	None
Settings file saved to:
	/Applications/Ardublockly.app/Contents/ServerCompilerSettings.ini

2016-11-28T16:46:15.909Z - info: [ArdublocklySrv] 
Arduino Board set to:
	Nano
Settings file saved to:
[... 137 lines cut ...]

Launching the HTTP service...

Provided Serial Port is not valid: !!!
	no_ports
Default Serial Port set:
	None
Settings file saved to:
	/Applications/Ardublockly.app/Contents/ServerCompilerSettings.ini

2016-11-28T17:49:46.237Z - info: [ArdublocklySrv] 
Provided Serial Port is not valid: !!!
	no_ports
Default Serial Port set:
	None
Settings file saved to:
	/Applications/Ardublockly.app/Contents/ServerCompilerSettings.ini

2016-11-28T17:49:47.643Z - info: [ArdublocklySrv] 
Provided Serial Port is not valid: !!!
	no_ports
Default Serial Port set:
	None
Settings file saved to:
	/Applications/Ardublockly.app/Contents/ServerCompilerSettings.ini

carlosperate, 2016-11-28:
Have you modified parts of the source code to try to force a specific serial port? Ardublockly is unable to parse invalid ports and set's it to "None". So I am not quite sure what would happen after that point.
There is also an error thrown by the Arduino application itself (Ardublockly just logs it), which shouldn't really happen on a simple "sketch verification".
Could you tell me a bit more about your environment? Ardublockly version, any possible modifications, Arduino version and your /Applications/Ardublockly.app/Contents/ServerCompilerSettings.ini file?

rp10007, 2016-11-28:
I haven't modified anything -- I tried back in october to modify the server settings file to force a serial port, but it just got overwritten by the "none". (I don't know if there's a particular format I should have been using.)

This is with a release copy of 0.1.2 (Also tried the latest nightly build, same thing). on a mac running 10.11.6, with Arduino 1.6.10. 

(Oh, and I think the error thrown by the Arduino IDE was because I had been playing with a Pro Trinket and switched to the nano. The trinket has that temporary-usb-port thing going on that makes a lot of things fail.)

puttley, 2017-04-30:
Hello,
Just wondering if there has been any resolution on this issue? I am having the exact same issue with the Mac version and in critical need of a fix or work around. Much appreciated...

carlosperate, 2017-04-30:
No, sorry, I haven't had a chance to look into this. Do you have a log file that I could have a look?
Ideally if the problem is easy to replicate in your system it would be easier for me if you first delete your original log, open the Ardublockly application, and within Ardublockly then opened the settings. If there is any selectable port in the dropdown menu it would be useful if you could select one as well.

puttley, 2017-04-30:
Thank you for the quick response. I can definitely generate a clean log file for you. Regarding the port dropdown menu, that is actually the issue at hand. There are no ports listed in the menu even though I have a UNO device connected and it is visible in the Arduino IDE. I've run into this only with the MAC version. MAC OS is El Capitan.

carlosperate, 2017-04-30:
I'll have to have a proper look, but I think the issue might be related with internal changes done within El Capitan: http://forum.arduino.cc/index.php?topic=351999.0
If this is the case I am not quite sure if I'll be able to fix this, as the limited access I've got to a mac is on a system with Yosemite.

puttley, 2017-05-03:
Hi Carlos,
Here is the log file you requested. Thank you for posting the thread regarding El Capitan and Arduino. However, like I mentioned before, I am able to see the serial ports just fine in the Arduino IDE and program an UNO in the IDE. In addition, an Ardublockly program sketch will verify fine in Ardublockly and all of the settings in the preferences are able to be set with the exception of the COM port. The COM port field continues to say "none".

Log File: OS El Capitan 10.11.1 | Arduino IDE 1.8.2

2017-05-02T21:57:12.828Z - info: [ArdublocklyElec] Starting Ardublockly version: 0.1.2
2017-05-02T21:57:12.829Z - info: [ArdublocklyElec] Ardublockly root dir: /Applications/Ardublockly.app/Contents
2017-05-02T21:57:12.829Z - info: [ArdublocklyElec] OS detected: darwin
2017-05-02T21:57:12.910Z - info: [ServerMgr] Command: /Applications/Ardublockly.app/Contents/arduexec/server/start --findprojectroot --nobrowser
2017-05-02T21:57:13.076Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-05-02T21:57:13.121Z - info: [ArdublocklySrv] Running Python 3.5.2 (64 bit) on Darwin-15.6.0-x86_64-i386-64bit

======= Parsing Command line arguments =======

Parsed "--findprojectroot" flag. The ardublockly project root will be set as the server root directory.
Parsed "--nobrowser" flag. No browser will be opened.

======= Resolving server and project paths =======

Ardublockly root directory: /Applications/Ardublockly.app/Contents
Selected server root: /Applications/Ardublockly.app/Contents

======= Loading Settings =======

Settings file corrupted or not found in:
	 /Applications/Ardublockly.app/Contents/ServerCompilerSettings.ini

Settings will be set to the default values.

Final settings loaded:
	Compiler directory: None
	Arduino Board Key: Atmel atmega168pb Xplained mini
	Arduino Board Value: atmel:avr:atmega168pb_xplained_mini
	Serial Port Value: None
	Sketch Name: ArdublocklySketch
	Sketch Directory: /Applications/Ardublockly.app/Contents
	Load IDE option: open

Settings file saved to:
	/Applications/Ardublockly.app/Contents/ServerCompilerSettings.ini

2017-05-02T21:57:44.851Z - info: [ArdublocklySrv] 

======= Starting Server =======

Setting HTTP Server Document Root to: 
	/Applications/Ardublockly.app/Contents

Launching the HTTP service...

IDE options set to:
	Verify sketch
Settings file saved to:
	/Applications/Ardublockly.app/Contents/ServerCompilerSettings.ini

2017-05-02T21:58:15.436Z - info: [ArdublocklySrv] 
Compiler file in OS X located within the app bundle.

Compiler directory set to:
	/Applications/Arduino.app/Contents/MacOS/Arduino
Settings file saved to:
	/Applications/Ardublockly.app/Contents/ServerCompilerSettings.ini

2017-05-02T21:58:48.296Z - info: [ArdublocklySrv] The "something" = ['else'] parameter is not recognised!

Sketch directory set to:
	/Users/pitscord/Documents/Arduino
Settings file saved to:
	/Applications/Ardublockly.app/Contents/ServerCompilerSettings.ini

[... 90 lines cut ...]

Ardublockly root directory: /Applications/Ardublockly.app/Contents
Selected server root: /Applications/Ardublockly.app/Contents

======= Loading Settings =======

Settings loaded from:
	 /Applications/Ardublockly.app/Contents/ServerCompilerSettings.ini

Settings file Serial Port is not currently available:
	None
Default Serial Port set:
	None

Final settings loaded:
	Compiler directory: /Applications/Arduino.app/Contents/MacOS/Arduino
	Arduino Board Key: Uno
	Arduino Board Value: arduino:avr:uno
	Serial Port Value: None
	Sketch Name: ArdublocklySketch
	Sketch Directory: /Users/pitscord/Documents/Arduino
	Load IDE option: verify

Settings file saved to:
	/Applications/Ardublockly.app/Contents/ServerCompilerSettings.ini

carlosperate, 2017-05-06:
Thank you @puttley, could you try this build and see if it works?

https://s3-us-west-2.amazonaws.com/ardublockly-builds/mac/ardublockly_Darwin64_2017-05-06_16.36_serial_ports_1de2c.zip

puttley, 2017-05-07:
Hello Carlos.

Thank you for the latest build. When I tried it out, the app would not launch. The error message said that Ardublockly would not work with this version of OS X. I've attached two screen shots that show the message that I got when trying to run it, and the detailed report. However, I did find an interesting way to force it to launch. I dragged the framework folder from the ardublockly_v01.2_osx build into your new build folder and replaced. Replacing the framework folder made your latest build launch. The good news is that the serial port showed up. However, when trying to set the Arduino compiler location or the sketch directory in the preferences, I cannot get a dialog window to open. If I click on either of these two options, nothing happens and after about 10 seconds the message of "Ardublockly App not Running" comes up.

On another note, when working from the v01.2 build, the serial ports are shown if I launch it by executing start.py and run in a browser. In addition, once the app has been launch using "start.py", I can close it, then launch the desktop app and the serial ports show up in the desktop preferences menu. It seems like once the app is launched from the core using start.py, the desktop app will then work. But, the start.py has to run first - one time.

I'm really hoping the MAC desktop app can be fixed. Our end user application for MAC is very much dependent on it and would be much easier if we didn't need to install Python and run it using start.py.

Thank you.

[image]

[image]

carlosperate, 2017-05-07:
Thanks @puttley, I was aware of the issue with setting the Arduino compiler location or the sketch directory in the preferences, but between today and tomorrow I will be changing the way that feature works, so hopefully that shouldn't be a problem soon.
I'm glad to hear that the ports do appear in this build (without having to call the python scripts first), so the next thing would be to look into why that error appears when trying to open Ardublockly.

As I mentioned before I have very sporadic access to a mac (and most of the time is in Yosemite), so unfortunately this will be hard for me to analyse and fix. But thank you for testing and these screenshots, it helps quite a lot.

Since the Framework folder has quite a few symlinks, and just to make sure, could let me know how did you unzip the application? There might be problems if using the command line interface or some programs that do not respect these symlinks correctly, but it should be fine if unzipped from the default macOS unarchiver.

puttley, 2017-05-07:
Hi Carlos,

Glad to know you were aware of those issues in preferences and they can be fixed. I'm really not a mac developer, but our end application is needing to be mac compatible. Me not being a mac person could mean that I unzipped it wrong. What I did was simply copy the zip file to my mac desktop and double clicked to extract. Should I be doing something differet?

puttley, 2017-05-07:
Update:

I re-tried unzipping your latest build on my MAC. This time it worked - the app launches OK. I'm not quite sure what went wrong the first time. But, there is still the issue of not being able to open the dialog boxes when clicking in the compiler location and sketch folder fields in the preferences settings window. For now to test, I cracked open the ServerCompilerSettings.ini file and manually pasted the paths in. From there I was able to see serial ports and upload code without any issues. If we can get the compiler location and sketch directory fixed in the preferences settings window, that should do it!

Thank you very much..

carlosperate, 2017-05-07:
hi @puttley, I'm glad the application worked when unzipped again.
I have reworked the file and folder dialogs, so that should be working again (the application is now smaller and the the new dialogs should feel more native as well, so it should be a win-win).
If you'd like to test it this is the latest version from master: https://s3-us-west-2.amazonaws.com/ardublockly-builds/mac/ardublockly_Darwin64_2017-05-07_22.30_master_64c5a.zip

carlosperate, 2017-05-08:
Better to try this one, fixes a small issue when there is no serial ports available: https://s3-us-west-2.amazonaws.com/ardublockly-builds/mac/ardublockly_Darwin64_2017-05-08_00.20_master_d5322.zip

puttley, 2017-05-08:
Carlos - this is great. The preferences selection dialog boxes are working now. The only exception is that when you first try to select the Compiler location, the dialog box opens and you can select the location. But, the path doesn't show up in the Compiler location area in the preferences box. It turns the line underneath it a red color, but doesn't show the path. If you close the preferences menu, then re-open it, the compiler path shows up. Any idea on this one?

carlosperate, 2017-05-08:
Hi,
Try the link from my second message again, I fixed that last night again but only realised this morning I posted the wrong link (so if you clicked on the link from your email it'd be the old one, sorry about that):
https://s3-us-west-2.amazonaws.com/ardublockly-builds/mac/ardublockly_Darwin64_2017-05-08_04.48_master_dc34b.zip

puttley, 2017-05-08:
Hi Carlos,

I tried the updated link, but it seems to have not resolved the problem. When first opening the preferences and attempting to set the Arduino compiler location, after clicking arduino location in the opened dialog box, the red line appears in the preferences but no information is shown on the line.
[image]

If I click return, then re-open the preferences, the compiler location appears.

Also, during testing - sometimes the "Ardublockly App not Running" message appears in the background even though the code successfully uploads. It's not predicatble - just sporatic. Any idea why that would happen?

Thank you.

carlosperate, 2017-05-08:
That's odd the issue you showed in the screenshot should have been fixed in
that build. Could you delete the current log, try to set the compiler path
and send the produced log with that issue?

I haven't really compiled sketches very often with the recent changes, I'll
have to try compiling several times in a row and see if I can replicate
that behaviour, it's a bit hard to debug some of those sporadic issues.

puttley, 2017-05-08:
Yes, I'll send that log to you shortly. Thank you.

puttley, 2017-05-08:
Hi Carlos,

Here is the log file.

Thank you.

[image]

carlosperate, 2017-05-09:
Hi @puttley,

Thanks for sending the log, I forgot that at the moment any console.log or console.error messages sent from the front end was not really being captured in the ardublockly.log file yet. I've updated Ardublockly to capture these log messages as well, so hopefully that way I could see where the problem might be happening once the server sends its reponse back to the front end.
Could you try again with this version? https://s3-us-west-2.amazonaws.com/ardublockly-builds/mac/ardublockly_Darwin64_2017-05-09_12.57_master_fe1a2.zip

puttley, 2017-05-09:
Thanks Carlos,

I re-tried and am getting the same behavior. The compiler path doesn't appear on the line in the preferences when you select the location. But, it will show up after you close the preferences, then re-open. I also checked the serverCompilerSettings.ini file and it is writing correctly to that file.

Thanks.

[image]

carlosperate, 2017-05-10:
Hi @puttley,
Thank you so much for your help with the testing so far, not having proper access to the OS has caused me to debug in the dark, so sorry for all the different builds trying to get it running.
I think I've tracked that the problem, could you test this build? https://s3-us-west-2.amazonaws.com/ardublockly-builds/mac/ardublockly_Darwin64_2017-05-10_00.47_set_compiler_osx_beecf.zip

puttley, 2017-05-10:
Hi Carlos - Problem Solved! Thank you very much!

carlosperate, 2017-05-10:
That's great to hear, thanks for all the help testing!

closed 2017-05-10 by carlosperate