carlosperate/ardublockly #122: Uploading arduino sketch with ardublockly stand-along version doesn't work porperly

skytrace2016, 2016-10-14:
Hi Carlos:
    I have just find your excellent project to program Arduino using Blockly  and Already compiled the whole windows stand-along project based on your wiki,But unfortunately, I found the same problem happened to me  relate to #59 and #27. The Uploading icon spinning and can not stop until I closed the program windows, And  have test the environment both on windows 8.1 and windows 10 with Arduino 1.6.5 1.6.7 1.6.12. I definitely sure the compiler setting is right ,Setting to the debug.exe and the right com port.But no expecting result comes out.  So  please help me find a solution,thank you so much.

carlosperate, 2016-10-15:
Hi @skytrace2016,
Sorry you are having issues with Ardublockly. Does this happen every time, or just occasionally?
If instead of compiling the program you chose to "open the sketch in the IDE", does the Arduino IDE open correctly?
Also, it could be quite useful if you could upload the ardublockly.log file somewhere to see if there are any erros capture there.

mrobotby, 2017-01-12:
I have the same problem. Arduino board receives Sketch, but no output in the window o Arduino IDE Output. Here is my log. Do you know how to fix this bug?

carlosperate, 2017-05-06:
Closing due to inactivity from original poster, please do feel free to reopen if you are still experiencing this issue.

I assume @mrobotby issue is already covered in https://github.com/carlosperate/ardublockly/issues/139 and the pending #141.

closed 2017-05-06 by carlosperate