carlosperate/ardublockly #123: More (older) arduino processor support

Lapjeskater, 2016-10-20:
Hey Guys,

To use the windows desktop ardublockly together with our teaching robots we need support voor a bit older arduino version. Its an Arduino Nano with ATmega168 processor, the arduino software is able to compile for it, so i would figure its just a setting in this program. Can you add it?

Thanks a lot!

carlosperate, 2016-10-28:
I've just added support for the Nano 168 board. If you try the latest builds uploaded into http://ardublockly-builds.s3-website-us-west-2.amazonaws.com/index.html it should be able to work. Let me know how you find it :)

carlosperate, 2016-11-03:
Hi @Lapjeskater, have you been able to test this board?

Lapjeskater, 2016-11-15:
Hey @carlosperate, I haven't been able to get opening in arduino or uploading to the board working yet, so at the moment im just coping the code. Ill reply again when i have some more details.

Greets

carlosperate, 2017-04-24:
The board support has been added in https://github.com/carlosperate/ardublockly/commit/d07a34549dd0717d5c9024fa74d98622c86481a0, since it's been a few months I assume this is working as it is. Please do reopen the issue if it doesn't.

closed 2017-04-24 by carlosperate