carlosperate/ardublockly #124: FieldInstance reload changes field name

bmcage, 2016-10-21:
Working with the new fieldinstance for stepper. 
Add two steppers to the workspace: Stepper1 and Stepper2.
Now add two move statements, one for Stepper1 and one for Stepper2. 

Reload the webpage. 
Stepper2 move statement is renamed to one of Stepper1. It seems to be due to the unique bool parameter:

```
      } else {
        // Pick an existing name from the workspace if any exists
        var existingName =
            Blockly.Instances.getAnyInstanceOf(this.instanceType_ , workspace);
        if (existingName) this.setValue(existingName);
      }
```
So for movement statements the first existingName is selected on rename, as unique == false for these fieldinstances.
It seems that if a name is present that is not default name, then the set name must be kept? Or some other way to fix this?

bmcage, 2016-10-21:
Possible solution:
```

      } else {
        // Pick an existing name from the workspace if needed and any exists
        var instanceList =
        Blockly.Instances.allInstancesOf(this.instanceType_,
                                         this.sourceBlock_.workspace);
        if (instanceList.indexOf(this.getValue()) == -1) {
          var existingName =
              Blockly.Instances.getAnyInstanceOf(this.instanceType_ , workspace);
          if (existingName) this.setValue(existingName);
        }
      }
```

SebCanet, 2017-06-07:
Hello,
did you find a way ? I still have the pb and do not have any idea to resolve it...