carlosperate/ardublockly #125: Servo should work on digital pins

bmcage, 2016-10-23:
The servo part has the following
``` 
        .appendField(new Blockly.FieldDropdown(
Blockly.Arduino.Boards.selected.pwmPins), 'SERVO_PIN');
```
So it forces PWM pins. However, servo can run on all digital pins, see https://www.arduino.cc/en/Reference/Servo where it says: _The Servo library supports up to 12 motors on most Arduino boards_

carlosperate, 2016-10-27:
Yeah, this has been on my todo list for a little while (https://github.com/carlosperate/ardublockly/issues/48). The main reason it hasn't been updated yet (as it is a rather quick change) it's because I need to think about how to disable pwm on those two pins. So I wanted to be able to dynamically create the pin dropdown in order to correctly apply that rule.
Think of it, it's probably better if I change it to use all the available pins and then update the fielDropdowns appropriately.

bmcage, 2016-10-27:
You are right. The use case of blockly is mostly learners, who probably will not run into this problem, I changed my servo: https://github.com/bmcage/ardublockly/commit/7c8029bfd1b2ce43905bc11d00fdf4f2060f22b1

But a setting in board that if servo include is present to change the possible fielddropdown, might be possible. I also have a board with an IR receiver switch on it, for that to work digital pins 0 and 1 can no longer be used. So a general way: rules to check to update the pin lists of boards.

carlosperate, 2016-10-27:
> You are right. The use case of blockly is mostly learners, who probably will not run into this problem, I changed my servo: bmcage@7c8029b

I would say that because it is aimed at learners, this is a bigger problem. If they use a servo, and an analogeWrite on pin 9, and the analogeWrite doesn't work there is absolutely no indication as to why it doesn't work and they will assume something is wrong with the code or the circuit.

Thanks for bmcage@7c8029b for reference. Looking at the documentation it sounds like it's able to use any pin, the only limitation being the number of servo pins attached at the same time. I tried compiling with pins A0 to A5 and didn't give any errors. Have you been able to try the servo on non-pwm pins including the analogue ones?

carlosperate, 2016-10-27:
Nevermind, of course `Blockly.Arduino.Boards.selected.digitalPins` including digital and analogue pins, doh!