carlosperate/ardublockly #126: infinite recursion in onchange with warning text in updateFields

bmcage, 2016-10-30:
It is possible to trigger an infinite recursion in the onchange method.
This is due to updateFields setting a warning text when changing a board.

How to trigger it:
Set board to Mega
Add digital pin read with pin selected to eg 24
Click on the warning indicator, so that the popup opens with the warning text
now change the board to uno --> the GUI hangs 

Probably some other interaction might be needed also, as sometimes the hanging seems to be only after doing this a second time or open/close of the warning

Analysis seems to indicate that setting a warning text in updateFields triggers an onchange call. For some reason this onchange keeps on being called. 

I hit this a lot in my version, as board can be changed with a component block, and I change test during testing. A solution seems to be to have updateFields change the dropdown fields only, and have the onchange of the block generate the warning instead. Disadvantage is that the test runs always then. Perhaps updateFields can set a flag, then onchange only tests as long as the flag is set, when no error the onchange unsets the flag.

carlosperate, 2016-11-03:
Hi @bmcage,
Just wanted to apologise for the delay on responding, I'm a bit busy at the moment, but this (and the other issues you've raised) is definitely in my shortlist of things to look up as soon as I get a chance.

bmcage, 2016-11-03:
No worries, we have an allbot camp this week, and I needed to add components for that. I could only hack some solutions for the issues I found on my branch, I will also only have time later to fully look into it to see what could be better/good solutions. 
FYI: See doc camp on https://docs.google.com/document/d/15rfbj0FlHattHjDwIhFxdCx6LSguy1hS9LRMJ4qW2J0/pub and images on facebook https://www.facebook.com/IngegnoMakerSpace/ 
Velleman made a scratch method to program the AllBot this week (http://manuals.velleman.eu/article.php?id=457 ), but one of the kids tried that and says our blockly is better :-) 
Unfortunately codebender.cc shuts down, so I need to find another solution to work from chromebooks.

carlosperate, 2017-05-06:
Sorry @bmcage I completely forgot about this one and it's taken so incredibly long to get around fixing it.
I looked into this for https://github.com/carlosperate/ardublockly/issues/165 and should be fixed in https://github.com/carlosperate/ardublockly/commit/599da1eefa6ae3cbf69ff5cacb784d19d439c94b.
Please reopen is it's not fixed or there is more to it.

closed 2017-05-06 by carlosperate