carlosperate/ardublockly #128: windows v.0.1.2 error when trying to run ardublockly_run.bat

Simsmzzms, 2016-11-02:
Downloaded most recent v. of ardublockly, but when trying to run the "pre-packaged" executable I repeatedly encounter an error. 

Restarting and all basic fixes have been tried. After I close the error message, the program runs but all of the GUI is does not appear.

The core version will open in Chrome but the GUI is totally unresponsive. Have I forgotten to do something?....I thought this file after extraction was meant to be self-enclosed.

Any suggestions would be greatly appreciated, thanks!

Simsmzzms, 2016-11-02:
It shows up as a JavaScript error. I tried re-extracting, and extracting the folder within the arduexec folder. As well as replacing the arduexec/server folder from the last build. Still no dice.

Simsmzzms, 2016-11-02:
Even the last build v.1.1 isnt running. Please help.

carlosperate, 2016-11-02:
If you can copy the error here I can then try to investigate?
I would also need the contents of the ardublockly.log file, so if you could upload it somewhere or copy it here it would be quite useful.

Simsmzzms, 2016-11-02:
Hi carlos,

Below is the info you asked for. Hopefully this clarifies the issues some.

Thank you so so much!

[ardublockly.txt](https://github.com/carlosperate/ardublockly/files/566617/ardublockly.txt)
[image]
[image]

carlosperate, 2016-11-02:
Could you confirm that the arduexec\server\start.exe file exists and that
you can execute it successfully? It should open a command line window with
some information and automatically open your default browser pointing to
the Ardublockly web interface. If it does open correctly, could you copy
and paste the contents of the console here as well?

Thanks,
Carlos

Simsmzzms, 2016-11-02:
@ C:\ardublockly\arduexec\server                  no start.exe file exists

Within that there is a folder called library which i've extracted. Is the start.exe file located in here? 

Otherwise what are your suggestions? Thanks again.

Simsmzzms, 2016-11-02:
[image]

Simsmzzms, 2016-11-02:
what scripts should be contained? I can try and make my own

carlosperate, 2016-11-02:
These are the contents from the zip file downloaded from the github releases link:

[image]

There should not be any need to rebuild the python server. Could you try to download the zip file and try again?

Simsmzzms, 2016-11-03:
Hi Carlos, sorry for the delay, I've done what you said and have managed to locate/execute the start.exe within the arduexec\server folder. Below are the contents of the console. Once again thank you so much, I'm excited to start utilizing this implementation.

[image]

Simsmzzms, 2016-11-03:
No need to further assist, everything is working wonderfully at this point. Its great!

closed 2016-11-03 by Simsmzzms

carlosperate, 2016-11-03:
Glad you git it working @Simsmzzms!