carlosperate/ardublockly #129: MAC copy/paste

Simsmzzms, 2016-11-03:
I've decided to also put this on a couple macs without and issues. Only thing that I have noticed is that when copying and pasting with command C/V there is a tendency to duplicate the paste. Not sure if this is something you are aware of but just figured i'd give you a heads up. Not a huge issue just wanted to make you aware.

carlosperate, 2016-11-03:
I've got a suspicion of where this problem might be coming from. Is it only on the macOs version where you experience this?
I'll look into this when I get some free time, hopefully during the weekend.
In the meantime, you could use the edit menu from the menubar at the top, I think that should avoid the duplication.

Simsmzzms, 2016-11-03:
Yeah its only duplicating on the macOsx version. The windows copy/pastes like normal. No rush whatsoever, just a comment more for transparency.

carlosperate, 2016-11-03:
Reopening as this is an issue i need to look into ;)