carlosperate/ardublockly #130: Grove Ultrasonic and Light Sensor Support

Simsmzzms, 2016-11-03:
Hi Carlos,

I know i've been bombarding you with questions as of late but I have just one more.

We are working to utilize Ardublockly in the classroom and one of the major lessons taught utilizes Ultrasonic Rangers and Light Sensors.

I am just wondering if there is a way to use them that I've missed or is this something you are planning on developing in the future? So far Ardublockly has worked like a dream so thank you.

carlosperate, 2017-01-22:
Hi @Simsmzzms , did you end up making new blocks for the ultrasonic and light sensors?

carlosperate, 2017-04-18:
Closing this due to inactivity, feel free to reopen if you have more questions or notes regarding this specific new blocks.

closed 2017-04-18 by carlosperate

ucakmuhgokhan, 2019-01-03:
Dear all,

Did you implement new block for ultrasonic sensor? I need to use HC SR04 with ArduBlockly, but there is no any block and pinq library.