carlosperate/ardublockly #131: iOS: Periodic infinite scrolling of Blockly window.

ELundby45, 2016-11-18:
This issue might be unique to my project. But I wanted to know if you have seen this before. 

Every now and then when loading Ardublockly and often when editing a number field; The blockly code window just starts growing and scrolling on iOS. 

I found a temporary solution by disabling `resizeBlocklyWorkspace` when editing a field. But I can't seem to figure out a solution to the seemingly random scrolling that occurs sometimes on load. 

Haven't seen this on any other platforms.

carlosperate, 2016-11-20:
Hi Eric,

Unfortunately I cannot say I have tested extensively on iOS, so I haven't really seen this specific issue myself. I remember a while ago there was some fix applied to the workspace size calculation having some type of issue, but I had a look around and cannot really find the commit.
Might be worth trying it with the Blockly demos website, see if you can replicate it in upstream Blockly.

ELundby45, 2016-11-21:
Thanks Carlos. I'm going to close this issue as I don't think it's a problem with ardublockly.

closed 2016-11-21 by ELundby45