carlosperate/ardublockly #133: Include scratch blocks port

rasmushaglund, 2016-12-08:
I'm interested in porting ardublockly to scratch blocks (https://github.com/LLK/scratch-blocks) and create some simplifying blocks that abstract some of the common complexity. The goal is to make it usable by children and act as a way of learning to code. 

Would this task fit in the project, or is it better to fork the project?

carlosperate, 2016-12-08:
As far as I understand the code generator has been removed from Scratch Blocks, as that is taken care by the Scract VM, were you planning to have something following that model or reapply the generator to the blocks?

rasmushaglund, 2016-12-22:
I think the easiest way would be to use the same generators as ardublockly. Sure it would be interesting to eventually add some kind of emulator, but since the process of uploading the code to the arduino to see the results is kind of easy you'll get quick feedback anyway.

SebCanet, 2016-12-22:
Hi,
that's a very good idea, so good that someone has already created a project like this : [http://kittenbot.cc/](http://kittenbot.cc/)

Hope we could work all together on great project for kids.

carlosperate, 2016-12-22:
Using Scracth VM doesn't really mean emulating an Arduino, but to implement the Arduino code generation in the "VM project", to then be passed to the IDE the same way Ardublockly does at the moment. With scratch there is a separation of concerns, so the blocks project is only responsible to provide the visual aspect, the Scrtach VM tracks the blocks and generates the AST, Scratch Render does the animation, Scratch GUI binds all the front end together, etc

I've been keeping an eye on Kittenbot, but unfortunately they haven't open sourced any of the Scratch changes, they have some repositories on GitHub only with readme files about the project.

rasmushaglund, 2016-12-22:
Yeah, I was thinking that Scratch VM could be used to "emulate" some parts of an Arduino in the browser. For example, instead of animating x/y positions of a sprite you could animate the light of a led and have it display inside the browser. This is secondary though, the main point is just replacing the visuals with scratch blocks.

RoboErikG, 2017-11-02:
Quick note from the Blockly team, the top level language generators have been added back in to Scratch Blocks. There just aren't any block generators included for Scratch's blocks. Otherwise, code generation should still work exactly the same as in Blockly (file an issue if it doesn't!). =)