carlosperate/ardublockly #135: Blinking window without any blocks after build | Server is not running

mrobotby, 2017-01-06:
Hi. I have built you program successfully, but when I run program only blinking windows appears without anything:( Do you know where is the problem? I use Python 2.7.
2017-01-06T16:46:55.333Z - info: [ArdublocklyElec] Starting Ardublockly version: 0.1.2
2017-01-06T16:46:55.334Z - info: [ArdublocklyElec] Ardublockly root dir: C:\ardublockly_Windows64_2017-01-06_19.43
2017-01-06T16:46:55.334Z - info: [ArdublocklyElec] OS detected: win32
2017-01-06T16:46:55.440Z - info: [ServerMgr] Command: C:\ardublockly_Windows64_2017-01-06_19.43\arduexec\server\start.exe --findprojectroot --nobrowser
2017-01-06T16:46:55.580Z - error: [ArdublocklySrv] Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\py2exe\boot_common.py", line 92, in <module>
    import linecache
ImportError: No module named linecache
Traceback (most recent call last):
  File "start.py", line 10, in <module>
    from __future__ import unicode_literals, absolute_import
ImportError: No module named __future__

2017-01-06T16:46:55.585Z - info: [ArdublocklySrv] Process exited with code 255
2017-01-06T16:46:59.039Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-06T16:47:00.790Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-06T16:47:01.472Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-06T16:47:03.197Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-06T16:47:03.878Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-06T16:47:05.606Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-06T16:47:06.289Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-06T16:47:08.011Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-06T16:47:08.689Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-06T16:47:10.420Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-06T16:47:11.091Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-06T16:47:12.821Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-06T16:47:13.494Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-06T16:47:15.220Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-06T16:47:15.903Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-06T16:47:17.622Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-06T16:47:18.308Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-06T16:47:20.022Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-06T16:47:20.715Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-06T16:47:22.419Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-06T16:47:23.116Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-06T16:47:24.909Z - info: [ArdublocklyElec] Starting Ardublockly version: 0.1.2
2017-01-06T16:47:24.911Z - info: [ArdublocklyElec] Ardublockly root dir: C:\ardublockly_Windows64_2017-01-06_19.43
2017-01-06T16:47:24.911Z - info: [ArdublocklyElec] OS detected: win32
2017-01-06T16:47:24.948Z - info: [ServerMgr] Command: C:\ardublockly_Windows64_2017-01-06_19.43\arduexec\server\start.exe --findprojectroot --nobrowser
2017-01-06T16:47:25.025Z - error: [ArdublocklySrv] Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\py2exe\boot_common.py", line 92, in <module>
    import linecache
ImportError: No module named linecache
Traceback (most recent call last):
  File "start.py", line 10, in <module>
    from __future__ import unicode_literals, absolute_import
ImportError: No module named __future__

2017-01-06T16:47:25.053Z - info: [ArdublocklySrv] Process exited with code 255
[image]

mrobotby, 2017-01-09:
I found that this problem I can avoid if before I start start.py file. Then I close web interface. Then I open ardublockly_run to run desktop application and all is working well. But when I try to start desktop application without starting start.py -nothing working, apears problem which I described above.

MarceloFariaz, 2017-01-10:
I have the same problem. My ardublockly.log below:

2017-01-10T19:24:58.556Z - info: [ArdublocklyElec] Starting Ardublockly version: 0.1.2
2017-01-10T19:24:58.558Z - info: [ArdublocklyElec] Ardublockly root dir: C:\...\ardublockly_Windows32_2017-01-10_12.32
2017-01-10T19:24:58.558Z - info: [ArdublocklyElec] OS detected: win32

2017-01-10T19:24:58.578Z - info: [ServerMgr] Command: C:\...\ardublockly_Windows32_2017-01-10_12.32\arduexec\server\start.exe --findprojectroot --nobrowser

2017-01-10T19:24:58.645Z - error: [ArdublocklySrv] Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\py2exe\boot_common.py", line 92, in <module>
    import linecache
ImportError: No module named linecache
Traceback (most recent call last):
  File "start.py", line 10, in <module>
ImportError: No module named __future__

2017-01-10T19:24:58.649Z - info: [ArdublocklySrv] Process exited with code 255

2017-01-10T19:25:01.091Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-10T19:25:02.841Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-10T19:25:03.510Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-10T19:25:04.028Z - warn: [ArdublocklyElec] Page failed to load (0). The server is probably not yet running. Trying again in 200 ms.

The compiled version works fine, but the distributable form does not.

mrobotby, 2017-01-10:
Help us, please!

carlosperate, 2017-01-11:
Hi,

When running the build scripts it's better to do it from the command line. I haven't really IDLE much, but I would that some of the error you are getting on those screenshots are because of that.

About running the start.py and desktop application together: You can only run one or the other. If there has been errors in the application sometimes the server might be left behind, so I would recommend checking your task manager to kill any outstanding instances.
If the desktop application does not work, and then you run start.py and the desktop application starts working then there is a problem starting the server.

Could you try running the build scripts from the command line interface and see if the server has been built correctly? It should be in `arduexec\server\start.exe`.

carlosperate, 2017-01-11:
@MarceloFariaz Could you let me know how have you downloaded, built, or installer Ardublockly?

MarceloFariaz, 2017-01-11:
Hey, @carlosperate !

I followed step-by-step according to the Building Ardublockly documentation, from the download of GitHub to the final step: Packing all Ardublockly. I used the Windows Command Processor to run each command line as an administrator, all time.

**About my setup of the Ardublockly executable build dependencies:**
Git version 2.10.0.windows.1
Python 2.7.12
Node v4.5.0
Mkdocs, version 0.16.1
Py2exe v0.6.9
pyinstaller v3.2

**About the import error on my ardublockly.log:**
ImportError: No module named linecache
[image]
ImportError: No module named future
[image]

Thank you for your attention!=)

mrobotby, 2017-01-11:
Now is my turn to check @carlosperate recommendations:)

mrobotby, 2017-01-11:
All is working if you will build build.py with Python27, and other files you will build with Python34 from CMD! Thank you @carlosperate 
But... I think I will have enough questions, because I'm going to add my own blocks to the application.

MarceloFariaz, 2017-01-16:
I still can not solve my problem. I tried using both versions of Python27 and Python34, but I still have the same result. I think it's something related to the Python + py2exe configuration, but I still can not find anything that would help me solve this problem. =/

MarceloFariaz, 2017-01-18:
@mrobotby can you tell me exactly which versions of the Ardublockly executable build dependencie you got made it work?

About my setup of the Ardublockly executable build dependencies:
Git version 2.10.0.windows.1
Python 2.7.12
Node v4.5.0
Mkdocs, version 0.16.1
Py2exe v0.6.9
pyinstaller v3.2

MarceloFariaz, 2017-01-18:
@carlosperate could you look at this link: http://www.py2exe.org/index.cgi/ProblemsToBeFixed ? I did not quite understand, but apparently it has to do with the problem I am facing. Thanks for your help!

carlosperate, 2017-01-19:
@MarceloFariaz could you open a new issue summarising your problem? It'd make it a lot easier to track and figure out.

carlosperate, 2017-01-22:
>All is working if you will build build.py with Python27, and other files you will build with Python34 from CMD! Thank you @carlosperate

@mrobotby This issue seems different than the one you've mentioned in https://github.com/carlosperate/ardublockly/issues/142, is it safe to close this one and continue in a new one for that specific problem?

carlosperate, 2017-04-18:
Since the original issue has been addressed and a new one opened to track the other problem I'm closing this one.

closed 2017-04-18 by carlosperate