carlosperate/ardublockly #136: How to Add Russian Language

mrobotby, 2017-01-07:
Hi. I want to add Russian language, but it seems not working well( Because when I try to add Russian language, only ??????? characters appear instead of normal letters. Are you able to help me with this issue?
Thank you.

carlosperate, 2017-01-09:
Could you post a screenshot or a link to a repository with the issue? It would help me figure out what the problem could be.

mrobotby, 2017-01-09:
I just can't understand how to add new language. How to add new? Because in guide is not clear description.

mrobotby, 2017-01-09:
First I open blockly/msg directory. There is only one file messages.js
But in this directory no file blockly/msg/<language>.js How to create it?
Also I don't know how add Russian block to choose in menu of the application (front end).

mrobotby, 2017-01-10:
[image]
It looks like this. 1. I just added file ru.js to directory ardublockly\ardublockly\msg 

2. And added a line for front-end to choose Russian language in Settings menu here ardublockly_lang.js in directory ardublockly\ardublockly
3. Then I just run start.py (without any build of the program).
Where is the problem? Help me please.
ru.js file and ardublockly_lang.js are below

// ru.js
var Ardublockly = Ardublockly || {};
Ardublockly.LOCALISED_TEXT = {
  translationLanguage: "Русский",
  title: "MBlockly",
  blocks: "Blocks",
  /* Menu */
  open: "Открыть",
  save: "Сохранить",
  deleteAll: "Удалить все",
  settings: "Настройки",
  documentation: "Документация",
  reportBug: "Сообщить об ошибке",
  examples: "Примеры",
  /* Settings */
  compilerLocation: "Путь к компилятору",
  compilerLocationDefault: "Путь к компилятору неизвестен",
  sketchFolder: "Папка со скетчами",
  sketchFolderDefault: "Папка со скетчами неизвестна",
  arduinoBoard: "Ардуино плата",
  arduinoBoardDefault: "Ардуино плата неизвестная",
  comPort: "COM Порт",
  comPortDefault: "COM Порт неизвестен",
  defaultIdeButton: "Default IDE Button",
  defaultIdeButtonDefault: "IDE опции неизвестны",
  language: "Язык",
  languageDefault: "Язык неизвестен",
  sketchName: "Имя скетча",
  /* Arduino console output */
  arduinoOpMainTitle: "Arduino IDE вывод",
  arduinoOpWaiting: "Ждем IDE вывод...",
  arduinoOpUploadedTitle: "Скетч успешно загружен",
  arduinoOpVerifiedTitle: "Скетч успешно скомпилирован",
  arduinoOpOpenedTitle: "Скетч был открыт в Arduino IDE",
  arduinoOpOpenedBody: "Скетч должен быть загружен в Arduino IDE.",
  arduinoOpErrorUpVerTitle: "Компиляция или загрузка провалились",
  arduinoOpErrorSketchTitle: "Скетч не найден",
  arduinoOpErrorFlagTitle: "Неправильный command line аргумент",
  arduinoOpErrorFlagPrefTitle: "Preference passed to 'get-pref' flag does not exist",
  arduinoOpErrorIdeDirTitle: "Unable to find Arduino IDE",
  arduinoOpErrorIdeDirBody: "The compiler directory has not been set.<br>" +
                            "Please set it in the Settings.",
  arduinoOpErrorIdeOptionTitle: "What should we do with the Sketch?",
  arduinoOpErrorIdeOptionBody: "The launch IDE option has not been set.<br>" +
                               "Please select an IDE option in the Settings.",
  arduinoOpErrorIdePortTitle: "Serial Port unavailable",
  arduinoOpErrorIdePortBody: "The Serial Port is not accessible.<br>" +
                             "Please check if the Arduino is correctly connected to the PC and select the Serial Port in the Settings.",
  arduinoOpErrorIdeBoardTitle: "Unknown Arduino Board",
  arduinoOpErrorIdeBoardBody: "The Arduino Board has not been set.<br>" +
                              "Please select the appropriate Arduino Board from the settings.",
  /* Modals */
  noServerTitle: "Ardublockly app not running",
  noServerTitleBody: "<p>For all the Ardublockly features to be enabled, the Ardublockly desktop application must be running locally on your computer.</p>" +
                     "<p>If you are using an online version you will not be able to configure the settings nor load the blocks code into an Arduino.</p>" +
                     "<p>Installation instruction can be found in the <a href=\"https://github.com/carlosperate/ardublockly\">Ardublockly repository</a>.</p>" +
                     "<p>If you have Ardublockly already installed, make sure the application is running correctly.</p>",
  noServerNoLangBody: "If the Ardublockly application is not running the language cannot be fully changed.",
  addBlocksTitle: "Additional Blocks",
  /* Alerts */
  loadNewBlocksTitle: "Load new blocks?",
  loadNewBlocksBody: "Loading a new XML file will replace the current blocks from the workspace.<br>" +
[... 193 lines cut ...]
  if (search.length <= 1) {
    search = '?lang=' + newLang;
  } else if (search.match(/[?&]lang=[^&]*/)) {
    search = search.replace(/([?&]lang=)[^&]*/, '$1' + newLang);
  } else {
    search = search.replace(/\?/, '?lang=' + newLang + '&');
  }

  window.location = window.location.protocol + '//' +
      window.location.host + window.location.pathname + search;
};

/**
 * Finds and returns the requests string in the localised language.
 * If the translation is not returned, it fetches the original language string.
 * @param {string} stringId
 * @return {!string} The localised, original, or an empty string.
 */
Ardublockly.getLocalStr = function(stringId) {
  var text = Ardublockly.LOCALISED_TEXT[stringId];
  if (!text) {
    console.log('Localised text string ID "' + stringId + '" does not exists!');
  }
  return text || Ardublockly.DEFAULT_LANG_TEXT[stringId] || '';
};

carlosperate, 2017-01-11:
Is the russian text in the blocks rendering correctly? I am having issues with the rendering of your ru.js file even on a normal text editor, maybe there is problems with the file encoding? I assume this is not how it is normally displayed on your environment:

[image]

mrobotby, 2017-01-11:
BIG Thanks @carlosperate . All is working correctly when I changed ANSI to UTF-8.

closed 2017-01-13 by mrobotby

carlosperate, 2017-01-16:
Glad you got it working. If you'd like to submit a Pull Request it would be great to have Russian as an available language!