carlosperate/ardublockly #137: Help me please create new category Sensors and blocks for this category

mrobotby, 2017-01-13:
Hi @carlosperate I'm using this guide https://github.com/carlosperate/ardublockly/wiki/Create-Blocks , but nothing working. I want to create category Sensors (like Motors, Logic, Loops categories). To this category I want to add blocks. But I even can't created category. Category is not appears in the list of categories. 
Here is filed which was modified to add category Sensors https://1drv.ms/u/s!ApxakNruUlXWplruS6ZtuGrvZmgv ardublockly_blockly.js
https://1drv.ms/u/s!ApxakNruUlXWpl3oAEBRrz76Fhd6 ardublockly_toolbox.js

MxBoud, 2017-01-13:
Hello, 
I'm not sure my information will provide help, I'm only beginner with Ardublockly, but have you modified index.html to add the path of your new code files? 
https://github.com/carlosperate/ardublockly/blob/master/ardublockly/index.html

carlosperate, 2017-01-22:
To add a new category you will have to modify the toolbox javascript file, and to translate it correctly you will also have to add the new entries into the javascript files in ` ardublockly/msg/` and update the `Ardublockly.updateToolboxLanguage()` function.

Hopefully this commit will be able to guide you: https://github.com/carlosperate/ardublockly/commit/1ad1ecb0e6bccb43c9c23f04ad7712985d9b3a4e

carlosperate, 2017-04-18:
I assume the original question has been answered, feel free to reopen this issue otherwise.

closed 2017-04-18 by carlosperate