carlosperate/ardublockly #139: Why Arduino IDE >1.6.0 can't work correctly with ArduBlockly?

mrobotby, 2017-01-15:
I have tested @carlosperate Desktop Blockly Application on the brand new machine Windows 10 x64. As a result, Arduino IDE >1.6.0 can't work correctly with ArduBlockly. The main problem, that is no output to ArduBlockly's Arduino output monitor. Only “Waiting for the IDE output...” Sketch uploads to the board, but no output.
How to solve this problem?
Here is my log. I only wanted to compile Sketch.
https://1drv.ms/u/s!ApxakNruUlXWpl7iVRv_qKqi-pw8

carlosperate, 2017-01-16:
Which version exactly of the Arduino IDE are you using?
It seems like the log is incomplete, it doesn't show any of the expected information when a sketch is uploaded or verified. Are you sure you've attached the right file?

mrobotby, 2017-01-16:
Versions 1.6.4 ;1.6.7; 1.6.8 ;1.8.0

carlosperate, 2017-01-16:
So when you click on upload code,  it says "waiting for the IDE output" and nothing else happens? It does not flash the Arduino (at least once) even if the application is stuck there?

The log you've sent only shows the information that would normally be logged when the application is opened and nothing else is done (the same thing as opening Ardublockly and closing it immediately). Could you run delete the log file and run a test again? Hopefully the new log should contain more information.
There is already a similar bug reported, but in those cases it was clear from the log, I'd like to check if we are looking at the same problem, or something different.

mrobotby, 2017-01-16:
Sketch uploads to board correctly, but no output, and to upload new sketch
I should close Program and open it again.

mrobotby, 2017-01-16:
On version 1.6.0 I can only see detailed information about uploading,
compilation in IDE Output of the Ardublockly Program.

mrobotby, 2017-03-07:
Hi. Has anyone the same problem with me? That Arduino 1.8.1 (for example) can't work with standard ArduBlockly program. No output in Ardublockly and sketch can't upload to the board.
I want to add Arduino 101 board support for Ardublockly, but I can't even upload a sketch to the Arduino Uno board, if I use standard Ardublockly and Arduino 1.8.1 IDE. 
@carlosperate can you upload a sketch to the Arduino board using Arduino 1.8.1 and standard Ardublockly program?

carlosperate, 2017-03-07:
I've just tried with Arduino 1.8.1 and it worked on my system. Of course I understand you still have this problem, but at least we can discard a bug within a specific version of the Arduino IDE.
Just in case this could also be part of https://github.com/carlosperate/ardublockly/issues/141, do you have the Arduino IDE in a non-English language?

mrobotby, 2017-03-08:
Yes, non English. I will try on English version.

mrobotby, 2017-03-08:
@carlosperate I can just open Arduino IDE using button Open in Arduino IDE.
But if I try to upload or compile sketch no output. If I try to upload
sketch, it works (no output in Ardublockly too) and I need to restart
Ardublockly program after this uploading.

8 марта 2017 г. 9:59 AM пользователь "Maksim Masalski" <
maxxliferobot@gmail.com> написал:

mrobotby, 2017-03-08:
@carlosperate IT IS WORKING NOW! I'M VERY HAPPY! 
I change language of my Arduino IDE from Russian to English. And now I can see output in Ardublockly. Now I can make support for Arduino 101 board!!!

carlosperate, 2017-03-08:
I'm glad we found what the underlying issue is, now is a matter to look into the proposed solution.
Since it looks this turned out to be the same issue as #141 I'll close this one and continue the work on the other.
Let me know if you think it is something else and I'll reopen it.

closed 2017-03-08 by carlosperate

helloyucsh, 2017-03-14:
I used Arduino UNO .  I met the same problem! It confused me all day ! After  change language of my Arduino IDE from Chinese  to English. Restart Arduino 1.81 IDE ( Windows 10 x64.) . It could also output in Ardublockly.  Thanks!