carlosperate/ardublockly #141: Wrong encoding in arduino output

microspace, 2017-01-16:
With russian Arduino IDE, there is wrong encoding in characters
[image]

carlosperate, 2017-01-16:
Thanks for reporting the issue. How are you running Ardublockly? Have you build it, are you running it directly from the command line? If so, are you using Python 2 or 3?

Do you have a repository I could clone and debug manually? that would help me figure out at which point the encoding is not correct.

mrobotby, 2017-01-16:
@microspace А какую Arduino IDE используешь? А то у меня если Arduino IDE новее 1.6.0, то не выводится ничего в окно вывода программы Ardublockly, хотя скетч в плату успешно вгружается.

microspace, 2017-01-16:
@mrobotby, my version is definetely > 1.6.0. I can't remember for sure, but I think it is 1.6.13

microspace, 2017-01-16:
@carlosperate 

I run it by cloning the repo and double clicking `start.py`. I'm using python 3.5 under Windows 10 64.
Tomorrow I'll send you the repo.

mrobotby, 2017-01-16:
Did you build Ardublockly by yourself or just installed desktop version of
Ardublockly?

carlosperate, 2017-01-16:
@microspace the python script prints out which version of python executes the script. This is definitely something that should be fixed for all cases, but maybe check if running Python 3 fixes this as a workaround (as Python3 uses unicode out of the box).

If one of you have a forked repository I could just clone it would also help me figure out where the problem might be caused.

mrobotby, 2017-01-16:
I forked it. But my skills in git are very poor. Say me what I should to
do, to let you download my repo.

microspace, 2017-01-16:
@carlosperate, I've installed desktop version

carlosperate, 2017-01-16:
@mrobotby You could just zip the folder and upload it to the same online service you've used before, I just need an easy way to test it on my computer.

@microspace You've downloaded the desktop version, but to execute it you double click on the `start.py` file, is that correct? In that case it should display some text either in the console, a new window, or the python idle. In any way, it should start displaying some text, and at the top there should be something like:
```
Running Python 2.7.10 (32 bit) on Windows-7-6.1.7601-SP1

======= Parsing Command line arguments =======
...
```

What does it say in yours for the Python version?

microspace, 2017-01-16:
I tried to reproduce issue on Linux, as Windows machine is not available fro now.
Here is output https://gist.github.com/microspace/77e4c03e65b9446a15c22936ef78391e
Archlinux 64 bit, Arduino IDE 1.6.17

carlosperate, 2017-01-23:
Right, so even running Python 3 has the same issue.
By any chance do you know the encoding of the Arduino IDE output? Up until now I assumed it was UTF, but maybe it is something different? which language is the Arduino IDE configured? Maybe I can debug in my PC by changing that.

microspace, 2017-01-23:
Arduino IDE is set to Russian language. 
Encoding of Arduino IDE in my case is WIN-1251

carlosperate, 2017-01-24:
Thanks for getting back to me. I haven't been able to find the time to look into this yet, but it's definitely in my ToDo list.

microspace, 2017-02-03:
I've found quick fix. In `actions.py`:

```
            #out = six.u(out)
            #error = six.u(error)
            import locale
            current_locale = locale.getdefaultlocale()[1]
            out = out.decode(current_locale)
            error = error.decode(current_locale)
```

This method doesn't work if user changed system default language in Arduino IDE
In this case we should read parse `editor.languages.current=` setting in `preferences.txt` of Arduino IDE

But I think this is rare case