carlosperate/ardublockly #142: Server is not running at packed version

MarceloFariaz, 2017-01-19:
# My console procedure(short version):

**C:\Users\RoboCore\Documents\git>python --version**
Python 2.7

**C:\Users\RoboCore\Documents\git>pip freeze**
backports-abc==0.5
backports.ssl-match-hostname==3.5.0.1
certifi==2016.9.26
click==6.7
Jinja2==2.9.4
livereload==2.5.1
Markdown==2.6.7
MarkupSafe==0.23
mkdocs==0.16.1
py2exe==0.6.9
PyYAML==3.12
singledispatch==3.4.0.3
six==1.10.0
tornado==4.4.2

**C:\Users\RoboCore\Documents\git>node --version**
v6.9.4

**C:\Users\RoboCore\Documents\git>npm --version**
3.10.10

**C:\Users\RoboCore\Documents\git>git version**
git version 2.11.0.windows.3**

**C:\Users\RoboCore\Documents\git>git clone https://github.com/carlosperate/ardublockly.git**

**C:\Users\RoboCore\Documents\git>cd ardublockly**

**C:\Users\RoboCore\Documents\git\ardublockly>git submodule update --init --recursive**

**C:\Users\RoboCore\Documents\git\ardublockly>cd blockly**

**C:\Users\RoboCore\Documents\git\ardublockly\blockly>python build.py**

**C:\Users\RoboCore\Documents\git\ardublockly\blockly>cd ../**

**C:\Users\RoboCore\Documents\git\ardublockly>python package\build_py2exe.py**

**C:\Users\RoboCore\Documents\git\ardublockly>cd package\electron**

**C:\Users\RoboCore\Documents\git\ardublockly\package\electron>npm install**

**C:\Users\RoboCore\Documents\git\ardublockly\package\electron>npm run release**

**C:\Users\RoboCore\Documents\git\ardublockly\package\electron>cd ../../**

**C:\Users\RoboCore\Documents\git\ardublockly>python package\build_docs.py**

**C:\Users\RoboCore\Documents\git\ardublockly>python package/pack_ardublockly.py**

# Results
## Unpacked version:
[image]

### Works fine:
[image]

### ardublockly.log(unpacked version):
2017-01-19T13:16:23.654Z - info: [ArdublocklyElec] Starting Ardublockly version: 0.1.2
2017-01-19T13:16:23.654Z - info: [ArdublocklyElec] Ardublockly root dir: C:\Users\RoboCore\Documents\git\ardublockly
2017-01-19T13:16:23.654Z - info: [ArdublocklyElec] OS detected: win32
2017-01-19T13:16:23.701Z - info: [ServerMgr] Command: C:\Users\RoboCore\Documents\git\ardublockly\arduexec\server\start.exe --findprojectroot --nobrowser
2017-01-19T13:16:25.018Z - info: [ArdublocklySrv] Running Python 2.7.0 (32 bit) on Windows-post2008Server-6.2.9200

======= Parsing Command line arguments =======

Parsed "--findprojectroot" flag. The ardublockly project root will be set as the server root directory.
Parsed "--nobrowser" flag. No browser will be opened.

======= Resolving server and project paths =======

Ardublockly root directory: C:\Users\RoboCore\Documents\git\ardublockly
Selected server root: C:\Users\RoboCore\Documents\git\ardublockly

======= Loading Settings =======

Settings loaded from:
	 C:\Users\RoboCore\Documents\git\ardublockly\ServerCompilerSettings.ini

The provided compiler path in the settings file is not valid:
	None
Default compiler path set:
	None

Settings file Serial Port is not currently available:
	None
Default Serial Port set:
	None

Final settings loaded:
	Compiler directory: None
	Arduino Board Key: Atmel atmega168pb Xplained mini
	Arduino Board Value: atmel:avr:atmega168pb_xplained_mini
	Serial Port Value: None
	Sketch Name: ArdublocklySketch
	Sketch Directory: C:\Users\RoboCore\Documents\git\ardublockly
	Load IDE option: open

Settings file saved to:
	C:\Users\RoboCore\Documents\git\ardublockly\ServerCompilerSettings.ini

2017-01-19T13:16:29.608Z - info: [ServerMgr] Server stopped.

## Packed version:
[image]

### Does not work:
[image]

### ardublockly.log(packed version):
2017-01-19T13:22:26.280Z - info: [ArdublocklyElec] Starting Ardublockly version: 0.1.2
2017-01-19T13:22:26.280Z - info: [ArdublocklyElec] Ardublockly root dir: C:\Users\RoboCore\Documents\git\ardublockly_Windows32_2017-01-19_10.42
2017-01-19T13:22:26.280Z - info: [ArdublocklyElec] OS detected: win32
2017-01-19T13:22:29.630Z - info: [ServerMgr] Command: C:\Users\RoboCore\Documents\git\ardublockly_Windows32_2017-01-19_10.42\arduexec\server\start.exe --findprojectroot --nobrowser
2017-01-19T13:22:31.458Z - error: [ArdublocklySrv] Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\py2exe\boot_common.py", line 92, in <module>
    import linecache
ImportError: No module named linecache
Traceback (most recent call last):
  File "start.py", line 10, in <module>
    from __future__ import unicode_literals, absolute_import
ImportError: No module named __future__

2017-01-19T13:22:31.599Z - info: [ArdublocklySrv] Process exited with code 255
2017-01-19T13:22:38.342Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-19T13:22:40.058Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-19T13:22:40.847Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-19T13:22:42.532Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
2017-01-19T13:22:43.314Z - warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.

# Observations:
## Python versions:
I already tried with the versions 2.7.0, 2.7.12 and 2.7.13. For the versions 2.7.0 and 2.7.12 the result is the same of above. For version 2.7.13 Another error occurred, but we'll leave that to another post. (For all versions, I tried for the two possible architectures: x32 and x64.)
I also tried using Python2.7 for the first and second step of the build instructions and Python3.4 for the other steps according to the #135 , but I was not successful either. The result is the same.

## Start start.py file before ardublockly_run.bat
> I found that this problem I can avoid if before I start start.py file. Then I close web interface. Then I open ardublockly_run to run desktop application and all is working well. But when I try to start desktop application without starting start.py -nothing working, apears problem which I described above. 

Same as #135 

## Operating System
I did the same procedure for windows 7 and 10 (both x64). The result is the same.

mrobotby, 2017-01-19:
Now I have the same problem again. BUT! Three Win10 x64 machines with Ardublockly packed version are working well. But one machine with Win10 x64 can't start the server and has similar with your problem!
I attached log file of wrong machine with Ardublockly which can't start server ardublockly (2) https://1drv.ms/u/s!ApxakNruUlXWpmSY97aV7TnqDyHd
and log file with Ardublockly which can work on this wrong machine https://1drv.ms/u/s!ApxakNruUlXWpmVIuP2GCvrLkUc2 (3)
It is like a magic. Three computers with Ardublockly work well, one can't work....
But if I try to run on this wrong machine older version of Ardublockly it is working well! MagiC!

MarceloFariaz, 2017-01-20:
@mrobotby what's your python 3.4 and 2.7 versions? You're build the first steps with the 2.7 version and the final steps with the 3.4 version, right?

carlosperate, 2017-01-22:
Hi Marcelo, thanks for opening up a new issue.

The issue you are experiencing is because the the server application is not running on your packed version, which is a bit weird since it seems to work correctly on your unpacked version.
The packaging script copies the ardublockly folder and cleans up a few things so that it is ready for distribution. Is there any reason you need to pack the Ardublockly application?

From the looks of it it seems like you are getting a new error in the packed version, when it should be running exactly the same executable.

```
2017-01-19T13:16:23.701Z - info: [ServerMgr] Command: C:\Users\RoboCore\Documents\git\ardublockly\arduexec\server\start.exe --findprojectroot --nobrowser
2017-01-19T13:16:25.018Z - info: [ArdublocklySrv] Running Python 2.7.0 (32 bit) on Windows-post2008Server-6.2.9200
```
vs
```
2017-01-19T13:22:29.630Z - info: [ServerMgr] Command: C:\Users\RoboCore\Documents\git\ardublockly_Windows32_2017-01-19_10.42\arduexec\server\start.exe --findprojectroot --nobrowser
2017-01-19T13:22:31.458Z - error: [ArdublocklySrv] Traceback (most recent call last):
File "C:\Python27\lib\site-packages\py2exe\boot_common.py", line 92, in
import linecache
ImportError: No module named linecache
Traceback (most recent call last):
File "start.py", line 10, in
from future import unicode_literals, absolute_import
ImportError: No module named future
```

This looks would normally be an error when creating the `server.exe` application from running `python package\build_py2exe.py`. Are you completely sure you have reproduce all the steps as you have described? If there is a problem with `server.exe` creation then the problem should be persistent in both the "packed" and "unpacked" versions.

If that is not the problem I would have to guess something might have gone wrong during the packaging of the script perhaps? Hard to say without the output from   `python package\build_py2exe.py` and `python package/pack_ardublockly.py`.
These two directories below should be identically, and have all the same files inside (specially library.zip, but any missing file would create problems):

```
C:\Users\RoboCore\Documents\git\ardublockly\arduexec\server\
```
comparted to
```
C:\Users\RoboCore\Documents\git\ardublockly_Windows32_2017-01-19_10.42\arduexec\server\
```

Could you check if those two are the same?

carlosperate, 2017-01-22:
@mrobotby Your problem, while might look similar, is probably caused by something completely difference (unicode support).
Could you open a new issue to be able to track this one as well? Having two different conversations in this thread will make things difficult to separate.
Taking in consideration the two logs show the server application crashing/closing, then it shouldn't be working in any of your computers, maybe you have got a different instance of the Ardublockly server? (either running the python script, or another instance of ardublockly.exe or server.exe)

MarceloFariaz, 2017-01-23:
Hi, @carlosperate ! Thanks for your attention!
As you said in your answer, I have reason to believe that the problem is occurring at the packing step.

## Answering your questions:

>Is there any reason you need to pack the Ardublockly application?

Yes! The unpacked version is much larger than the packed version.

>Are you completely sure you have reproduce all the steps as you have described? 

Yes, i am!

>Could you check if those two are the same?

No, they are not! I think we found the possible problem.. look at this:

[image]

>Hard to say without the output from python package\build_py2exe.py and python package/pack_ardublockly.py.

About these logs: 
[build_py2exe_py.txt](https://github.com/carlosperate/ardublockly/files/724692/build_py2exe_py.txt)
[pack_ardublockly_py.txt](https://github.com/carlosperate/ardublockly/files/724694/pack_ardublockly_py.txt)

What could be happening?

One strange thing is that by running the start.py of the compressed version, the server seems to have been created successfully, launching the application in my browser, see:

[image]

Or is not this strange?

carlosperate, 2017-01-24:
Running start.py executes a python script, so as long as you have python installed that should work straight from the master branch (all the python code is included in the repository).
When the build_py2exe.py script is executed, it uses py2exe to pack together the Python interpreter with the python code into an executable for your Operating System. That way the distributable Ardublockly application does not depend on Python being installed in the users' computers.

Something is not quite write when building the py2exe executable, it seems like instead of packaging all the compiled python code into a library file it is leaving it uncompressed. Then the packaging script deletes all these files, as by design there should not be any .pyc files laying around.

The version you have installed of py2exe seems quite old, I think this might be the difference. The Python2 version hasn't been updated for a long time, and the current settings produce a different output than it does with Python 3 (which uses a newer version of py2exe). I would need to go check the build script and see if there is anything I could do to force the library.zip file to be created, or to modify the packaging script to not remove the pyc files.

If you can't wait (it might take a little while before I can look into this) you have two options:
a) Install Python 3 and py2exe for python 3 (there is a wheel available in PyPi, so you can use pip) and then run the build script with that: `py -3 package\build_py2exe.py`
b) Modify the packaging script to **not** deleted compiled python code by commenting out these two lines of code: https://github.com/carlosperate/ardublockly/blob/master/package/pack_ardublockly.py#L313-L314

MarceloFariaz, 2017-01-25:
### It finally worked!
Following this step-by-step worked first:
> Install Python 3 and py2exe for python 3 (there is a wheel available in PyPi, so you can use pip) and then run the build script with that: py -3 package\build_py2exe.py

Thanks so much for this, @carlosperate ! =)

carlosperate, 2017-01-25:
Glad you got it working!
I'll leave this open, as I need to update the py2exe build step in Python2 to have the same type of output than Python 3.

carlosperate, 2017-05-06:
The build steps have now changed, and all platforms use PyInstaller, so closing this as the original issue was already resolved and the outstanding change in the documentation is no longer required.

closed 2017-05-06 by carlosperate