carlosperate/ardublockly #146: JavaScript error on running Linux compiled version

MarceloFariaz, 2017-02-03:
After succeeding to compile and compact Ardublockly for Windows, I am now trying to do the same for Linux and Mac.
After succeeding to compile and compact Ardublockly for Windows, I am now trying to do the same for Linux and Mac.
By executing all the steps of the documentation as I did for the Windows version, I was able to compile and compact without any problems. 
### However, at the time of execution, I got this error:
[image]

### About my build tools versions:
[image]

Could you help me with that?

carlosperate, 2017-02-03:
Looks like winston wasn't installed correctly, when doing the `npm install` step did you get any errors?

MarceloFariaz, 2017-02-06:
I had some warnings only. Look at the image below:
[image]

carlosperate, 2017-02-06:
And have you executed `npm run release` as well?

MarceloFariaz, 2017-02-06:
Yes, I did:
[image]

carlosperate, 2017-04-18:
Hi @MarceloFariaz, sorry I was able to follow up at the time, do you still have this problem?

carlosperate, 2017-05-06:
Closing due to lack of activity, please do reopen the issue if you are still experiencing this problem.

closed 2017-05-06 by carlosperate

MarceloFariaz, 2017-06-09:
Hi @carlosperate following the steps of this link https://github.com/carlosperate/ardublockly/blob/master/circle.yml the problem was solved! Thanks =)

carlosperate, 2017-06-10:
I'm glad you managed to resolve this problem, was there any specific step that was missing or not clear enough? It would be great to make sure the build instructions are correct.