carlosperate/ardublockly #147: TypeError: LoadLibrary() argument 1 must be string, not unicode

hhernandez92, 2017-02-06:
After following the procedures on https://github.com/carlosperate/ardublockly/wiki/Building-Ardublockly and trying to launch "start.py", an exception is thrown: "TypeError: LoadLibrary() argument 1 must be string, not unicode".

hhernandez92, 2017-02-06:
I was able to solve the issue by changing the following lines of code in "list_ports_windows.py"

Line 83: 
From - setupapi = ctypes.windll.LoadLibrary("setupapi")
To - setupapi = ctypes.windll.LoadLibrary("setupapi".encode('ascii', 'ignore'))

and Line 113:
From - advapi32 = ctypes.windll.LoadLibrary("Advapi32")
To - advapi32 = ctypes.windll.LoadLibrary("Advapi32".encode('ascii', 'ignore'))

emil01, 2017-02-12:
@hhernandez92 Not exactly sure why the issue is but I resolved it by running start.py using python3

carlosperate, 2017-04-15:
Turns out to be a bug in Python 2.7.13, so the best options right now are to either user an earlier 2.7 version, or move to Python 3.
I'll close this here as no changes are required in the Ardublockly repository.

closed 2017-04-15 by carlosperate