carlosperate/ardublockly #152: Where to edit Big Round Icons -Upload, Compile, Open in Arduino IDE?

mrobotby, 2017-02-15:
I can't find file where to edit/delete this buttons. Does anyone know?
[image]

pteacher, 2017-02-18:
Im not sure, about reason you want to do that, but you can delete them in ardublockly\index.html file
lines 123-125
`          <a id="button_ide_left" class="waves-effect waves-light waves-circle btn-floating z-depth-1-half arduino_yellow"><i id="button_ide_left_icon" class="mdi-action-open-in-browser"></i></a>
          <a id="button_ide_middle" class="waves-effect waves-light waves-circle btn-floating z-depth-1-half arduino_teal"><i id="button_ide_middle_icon" class="mdi-navigation-check"></i></a>
          <a id="button_ide_large" class="waves-effect waves-light waves-circle btn-floating z-depth-1-half arduino_orange"><i id="button_ide_large_icon" class="mdi-av-play-arrow"></i></a>`

mrobotby, 2017-02-22:
谢谢 @pteacher ！

carlosperate, 2017-03-07:
Looks like @pteacher has been able to answer your question (thanks!), so I'll close this issue. Let me know if you still need any clarifications.

closed 2017-03-07 by carlosperate