carlosperate/ardublockly #158: Compilation and load does not work.

euclas, 2017-03-15:
Hi.

I installed ArduBlockly on:

C: \ Users \ Euclides \ Documents \ _Installs \ Blockly \ ardublockly

Already had the Arduino installed in:

C: \ Program Files \ Arduino

In ServerCompilerSettings.ini it looks like this:

[Arduino_IDE]
Arduino_exec_path = C: \ Program Files \ Arduino
Arduino_board = Mega
Arduino_serial_port = COM11

[Arduino_Sketch]
Sketch_name = ArdublocklySketch
Sketch_directory = C: \ Users \ Euclides \ Documents \ _Installs \ Blockly \ ardublockly

[Ardublockly]
Ide_load = open

By clicking the button to upload, nothing happens.

Does anyone know why?

Thank you.

Euclides Franco de Rezende
Brazil.

carlosperate, 2017-03-16:
Hi,
Make sure the `arduino_exec_path` points to the `arduino_debug.exe` file and not the installation directory.

euclas, 2017-03-16:
Hi,

It's the same directory:

C:\Program Files\Arduino

What about the ServerCompilerSettings.ini file? Is it in the same directory as ardublockly_run.bat? Or in the ardublockly.exe directory?

I need a Blockly converter for BASIC and I found your project.
I intend to use the BASIC language on a microprocessor (ESP8266-NodeMCU), which has an interpreter in operation and if I could do the Blockly generate source in BASIC language would be the solution.

I found people who were already interested in doing the conversion part of Blockly for the BASIC language:

Https://groups.google.com/forum/#!searchin/blockly/BASIC%7Csort:relevance/blockly/j21cRPTEEkk/Izu7rkOGJQAJ

He made it available on Github, but I do not know what state it is in:

Https://github.com/MikeDX/blockly-div

It would be missing just part of the source submission for ESP8266Basic (https://www.esp8266basic.com/). If the syntax is Ok.

There are differences between the Arduino and ESP8266 that need to be observed: pins, ports and utilities of each of the modules or hardware extensions to be used.

What's all this for?
Answer: So that the ESP8266 can be programmed via WiFi and not by cable, or any type of electrical connection, only by air.

I would like to know your opinion on this? Since you have much more experience than I do in this environment.

Thank you.

carlosperate, 2017-03-16:
Hi @euclas,

Just to confirm, your settings file has this exact entry?
```
arduino_exec_path = C:\Program Files\Arduino\arduino_debug.exe
```

And yes, the ServerCompilerSettings.ini file should be in the same directory as ardublockly_run.bat.

About the project in itself, I've never used BASIC before and don't really know anything about ESP8266Basic either, but as long as it has an interpreter built-in that could be feed the generated code then I think it should be possible.

euclas, 2017-03-16:
Hi Carlos.

Exactly:

[image: Imagem inline 1]

I would like to test at least the basics, in the Arduino IDE (COM11, and
Arduino Mega 2560) works correctly.

Thank you again

Euclides.

carlosperate, 2017-05-06:
HI @euclas, do you still have problems getting Ardublockly to use the Arduino IDE?