carlosperate/ardublockly #159: ardublockly to blocklyProcessing

dimpapadop, 2017-03-23:
Hallo, i am completely newbie to blockly and ardublockly and reading about it
I want to create a similar environment for Processing (not p5js that is pure javascript).

Since Processing is so similar to Arduino IDE i wanna help doing this (sketches will be run with command line in local installation of processing)

Can anyone help me to get started?
Is there a problem to do it?

thank you in advance

karlTH, 2017-04-01:
Hello,
may be take a look to this fork:
https://github.com/HackInvent/ardublockly

and change in ServerCompilerSettings.ini
the link to arduino.exe in processing.exe

SebCanet, 2017-04-01:
I think the idea is not only compiling in another IDE but do a "blockly to Processing code" graphical programming environnement.

dimpapadop, 2017-04-01:
@SebCanet  I agree, it's just a way better start, since Arduino IDE is based on Processing and it's so common in the parts that have already been implemented.

@karlTH  I'll try to check it. Processing-java.exe can run a sketch in Processing with cmd line, so i think that part'll be easier than i thought @ the beginning

SebCanet, 2017-04-01:
So you have to change the '\generator\arduino.js' file to change function you need.
And in '\generators' you have all the file used to translate a block in a text code.
Hope you'll success !!!

dimpapadop, 2017-04-01:
thank you @SebCanet  and @karlTH i'll keep you informed about the projects

SebCanet, 2017-04-01:
Some help maybe:
[http://blokke.bitballoon.com/](http://blokke.bitballoon.com/)
[https://forum.processing.org/two/discussion/12104/visual-sketch-editor-for-p5-js](https://forum.processing.org/two/discussion/12104/visual-sketch-editor-for-p5-js)

dimpapadop, 2017-04-01:
i checked them out. Quite a bit of blocks have already implemented.
Also working on adding a Serial Monitor to ardublockly that's for sure needed

carlosperate, 2017-04-02:
Thanks @SebCanet for helping out, I think you might have covered everything that I had in mind.

And also it is true that a serial monitor would be a great addition. I have a very crude version running in a local branch, but since I am also changing the back end a little bit I will probably add the serial monitor after that.

dimpapadop, 2017-04-02:
And with my tun thank you @carlosperate  for your wonderfull work on ardublockly.

Actually there are quite many ways in implementing a serial monitor. I'll let you know which we'll adopt and by our turn help ardublockly a bit

carlosperate, 2017-04-02:
In case you are interested, at the moment to integrate well with Ardublockly I'd be looking at a serial monitor in Python communicating to the front end using either websockets or socket.io.

karlTH, 2017-04-04:
it will be really interessting to have serial monitor in ardublockly

carlosperate, 2017-04-17:
Issue #29 is already opened to track adding a serial monitor, so since it looks all the other questions have been answered I'm closing this one.

closed 2017-04-17 by carlosperate

SebCanet, 2017-05-20:
@dimpapadop I found a possible source : [http://jforthblocks.appspot.com/static/code/index.html](http://jforthblocks.appspot.com/static/code/index.html)
To start: [http://jimmyscratchlab.blogspot.fr/2012/07/jforthblocks-processingjs-tutorial.html](http://jimmyscratchlab.blogspot.fr/2012/07/jforthblocks-processingjs-tutorial.html)

carlosperate, 2017-05-22:
Thanks @SebCanet! I`ll definitely check it out once I get a chance.