carlosperate/ardublockly #160: Ardublockly building problem

ioarvanit, 2017-03-26:
Hi. I have just downloaded the ardublockly-master and i can't build the windows executable.

I made some minor changes to blockly blocks and generators and succesfully build blockly. I am running localy ardublockly with python server and everything works fine.

The problem occurs when i try to build a new windows executable, I followed the instructions at https://github.com/carlosperate/ardublockly/wiki/Building-Ardublockly and when i run the build_py2exe.py script i get an empty arduexec folder and some traceback calls in the command line. (see image below)

I have python 2.7 and 3.6 installed, and i also installed py2exe, pyInstaller, MkDocs and node.js with electron. I have windows 10 32-bit. I would really appreciate any help on this!

[image]

carlosperate, 2017-04-02:
Hi @ioarvanit,

Apologies for the late reply. It looks like py2exe does not yet support python 3.6, and in fact is not yet compatible with python 3.5 either, so it looks like for the time being Python 3.4 will be required to build Ardublockly.

carlosperate, 2017-04-02:
I've also added a note in https://github.com/carlosperate/ardublockly/wiki/Building-Ardublockly to emphasise this incompatibility.

ioarvanit, 2017-04-06:
Thanks for the feedback.

carlosperate, 2017-04-17:
Closing this as the issue is an incompatibility from py2exe and the Python version.

closed 2017-04-17 by carlosperate