carlosperate/ardublockly #161: No ARD_PULSETIMEOUT_MS string

bmcage, 2017-03-30:
Why is there no pulse timeout  ms string? We have 

msg/messages.js:Blockly.Msg.ARD_PULSETIMEOUT = 'timeout after';
msg/messages.js:Blockly.Msg.ARD_PULSETIMEOUT_MS = '';

So the user must read the manual to see that timeout is expressed in microseconds. I would suggest:
msg/messages.js:Blockly.Msg.ARD_PULSETIMEOUT_MS = 'microseconds';
Or  μs

Motivation: had an error with user having code not working. I forgot it was in microsec, value was given in millisec. The interface just puts nothing...

carlosperate, 2017-04-02:
Thanks for pointing that out, looking at the git history that value was never populated in the "messages.js" file, and the string is also set to miliseconds in some of the translations, so I'll update that for the next update.

carlosperate, 2017-04-24:
I've updated the pulseIn() block in https://github.com/carlosperate/ardublockly/commit/14f03657eff4b6e9fe77a48dfab1d6da99412f32

Basically changed the block initialisation from from javascript to json, as it allows interpolation of strings and fields, which is quite important for the different grammar from translated languages.

The block has changed slightly from:
[image]

to:
[image]

Thanks for the report @bmcage!

closed 2017-04-24 by carlosperate