carlosperate/ardublockly #164: Storing function return value causes `Uncaught TypeError`

ELundby45, 2017-04-26:
Trying to store the return value of a function into a variable causes the following error:
```
blockly_compressed.js:1340 Uncaught TypeError: Cannot read property 'targetBlock' of null
    at Object.Blockly.Types.getChildBlockType (blockly_compressed.js:1340)
    at Blockly.BlockSvg.getVarType (blocks_compressed.js:153)
    at Function.Blockly.StaticTyping.getBlockVars (blockly_compressed.js:1344)
    at Blockly.StaticTyping.collectVarsWithTypes (blockly_compressed.js:1342)
    at Blockly.Generator.Blockly.Arduino.init (arduino_compressed.js:12)
    at Blockly.Generator.workspaceToCode (blockly_compressed.js:1324)
    at Object.Ardublockly.generateArduino (ardublockly_blockly.js:75)
    at Ardublockly.renderContent (ardublockly.js:527)
    at Blockly.WorkspaceSvg.Blockly.Workspace.fireChangeListener (blockly_compressed.js:883)
    at Blockly.Events.fireNow_ (blockly_compressed.js:1230)
```

Example:
[image]

carlosperate, 2017-04-30:
Thanks for the repor @ELundby45, I'll see if I can investigate this one between today and tomorrow.

carlosperate, 2017-05-03:
Should be fixed in 9ca434dc2221b6859e60a440226c371b638df512

ELundby45, 2017-05-03:
That did it. Thank you very much @carlosperate!

closed 2017-05-03 by ELundby45