carlosperate/ardublockly #165: loop in setWarningText

zhsvector, 2017-04-28:
I noticed in SerialPrint, in the absence of Setup,
When you open the warning bubble,
Blockly.Arduino['serial_print'] = function: set the warning message
Blockly.Arduino.reservePin = function: delete the warning message
so,Program loop in settings / delete,Page not responding.

carlosperate, 2017-04-30:
Hi @zhsvector ,
I'm sorry, but I am not quite sure I follow. If you are reporting an issue, could you provide the individual steps required to reproduce it?
Thanks,
Carlos

zhsvector, 2017-05-02:
Open ardublockly0.1.2
Place a "Serial print" block into the workspace
There will be a warning icon on it
Click it, there will be a bubble display WarningText
Now try to move the block ,It is no response
I try to follow it to find out why
So I see
Warning information is set in blockly\blocks\arduino\serial.js (line:114)
After,
in blockly\generators\arduino\serial.js(line:34),jump to Blockly.Arduino.reservePin() 
Blockly\generators\arduino.js (line:290) set warning information to null
So I guess
They will have been circulating calls, the performance of the JavaScript virtual machine depletion
 I'm not good at English,not sure I've made it clear:)

carlosperate, 2017-05-03:
Thank you for the clarification, I've been able to replicate and investigate.
You are correct there, we have some circular calls. 

When we open the warning box bubble it triggers a UI event, and that UI event triggers a code generation, this code generation then tries to update the warning text on the same block for a different reason.
When the warning text is updated Blockly hides and unhides the warning bubble (very quickly so it's not noticeable) and so it triggers a new UI event and it starts all over again.

carlosperate, 2017-05-03:
This should be fixed in https://github.com/carlosperate/ardublockly/commit/599da1eefa6ae3cbf69ff5cacb784d19d439c94b

carlosperate, 2017-05-06:
Closing as it's fixed in master, but please feel free to reopen if there are still issues in this area.

closed 2017-05-06 by carlosperate