carlosperate/ardublockly #169: Problem setting the compiler location (Run Ardublockly from source code - windows)

RubensQRZ, 2017-05-11:
Dear Carlos, 

I was testing the lates version of Ardublockly  (master source code) and I encountered a problem when I tried to set the compiler location.  

I am using Windows with Python V3.4.4 and I made the same tests using google Chrome and Mozilla Firevox  in 2 machines, one with windows 8 and other with windows 10.

The problema is:

When I click over the line to set the compiler location the window to select the location does not open (nor even behind  the browser) I when I return the focus to the browser the message “Ardublockly  app not running” appears. And at the python window we can read:

Traceback (most recent call last):
  File "C:\Ardublockly\ardublocklyserver\local-packages\bottle.py", line 862, in _handle
    return route.call(**args)
  File "C:\Ardublockly\ardublocklyserver\local-packages\bottle.py", line 1740, in wrapper
    rv = callback(*a, **ka)
  File "C:\Ardublockly\ardublocklyserver\server.py", line 337, in handler_settings_update_individual
    if set_value in new_value:
TypeError: 'in <string>' requires string as left operand, not NoneType

When I click over the line to set the Sketch Folder, nothing happens.

_________________________________________________________________________________
Running Python 3.4.4 (64 bit) on Windows-10-10.0.14393
Local packages: C:\Ardublockly\ardublocklyserver\local-packages

======= Parsing Command line arguments =======
No command line arguments found.

======= Resolving server and project paths =======
Ardublockly root directory:
        C:\Ardublockly
Current working directory set to:
        C:\Ardublockly
Selected server root:
        C:\Ardublockly
Selected server ip:
        localhost
Selected server port:
        8000

======= Loading Settings =======
Settings loaded from:
        C:\Ardublockly\ServerCompilerSettings.ini
The provided compiler path in the settings file is not valid:
        None
Default compiler path set:
        None
Final settings loaded:
        Compiler directory: None
        Arduino Board Key: Atmel atmega168pb Xplained mini
        Arduino Board Value: atmel:avr:atmega168pb_xplained_mini
        Serial Port Value: COM3
        Sketch Name: ArdublocklySketch
        Sketch Directory: C:\Ardublockly
        Load IDE option: open
Settings file saved to:
        C:\Ardublockly\ServerCompilerSettings.ini

======= Starting Server =======
Setting HTTP Server Document Root to:
        C:\Ardublockly
Launch Server:
Bottle v0.12.13 server starting up (using WaitressServer())...
Listening on http://localhost:8000/
Hit Ctrl-C to quit.

Serving on http://DESKTOP-OF0OQIA:8000
Serving on http://DESKTOP-OF0OQIA:8000
The provided compiler path is not valid !!!
         Please enter the location of the Arduino IDE executable
Default compiler path set:
        None
Settings file saved to:
        C:\Ardublockly\ServerCompilerSettings.ini
Traceback (most recent call last):
  File "C:\Ardublockly\ardublocklyserver\local-packages\bottle.py", line 862, in _handle
    return route.call(**args)
  File "C:\Ardublockly\ardublocklyserver\local-packages\bottle.py", line 1740, in wrapper
    rv = callback(*a, **ka)
  File "C:\Ardublockly\ardublocklyserver\server.py", line 337, in handler_settings_update_individual
    if set_value in new_value:
TypeError: 'in <string>' requires string as left operand, not NoneType

carlosperate, 2017-05-22:
Hi Rubens,

Thanks for reporting the issue, even if the behaviour has changed I should have protected a "None/null" type value sent to the server.
If it's okay with you, I'll reopen this issue to remind me to look into that, and ideally have a more helpful message printed in the log (instead of just the Python exception).

Did the input box at least turned colour red when this happened?

RubensQRZ, 2017-05-22:
Hi Carlos, 

I really dont remember if the input box turned colour red.

carlosperate, 2017-05-22:
No worries, I'll keep an eye during testing.

JulesTopart, 2017-12-26:
[image]

I have just fixed that on my side, I added a Browse button (So I've changed the bindclick call in Ardublockly.init() on Client side ) and make the browser appear on top of all opened windows (by changing the gui.py (or actions.py) on server side)