carlosperate/ardublockly #170: server doesn't recognize user blocks folder

40601672, 2017-05-14:
I added my custom block robotkit to root/blocks folder. I changed blocks_data.json but when I started a server and key in javascript:Ardublockly.openExtraCategoriesSelect() but I can only see Grove and Test option but not my robotkit. Any idea why?
Note 1. I inspect the loaded server page, it doesn't list new robotkit folder in blocks as well.
Note 2. I use uncompressed version and tried to recompile the server as instructed in https://github.com/carlosperate/ardublockly/wiki/Building-Ardublockly

carlosperate, 2017-05-22:
Hi @40601672,

It is important to note that this feature is still very very raw. It's quite high in my to-do list, so soon enought I should be able to give it some time to improve it.

Do you have a link to your projects? Is it somewhere on github where I could easily look at the modified json file?

40601672, 2017-05-23:
I understood, thanks for looking into. 

My repo can be found at https://github.com/40601672/ardublockly-robotkit