carlosperate/ardublockly #171: Adding the otto diy library to ardublockly

nickedens, 2017-05-19:
I run a non profit that teaches kids programming, electronics, 3d printer, etc. I think the otto DIY project robot from here http://otto.strikingly.com/ is absolutely amazing. However I really like the simplistic block interface that you have here. Is there any possibility of you incorporating the otto library into ardublockly? Being a very small non profit we don't have a ton of money but I would be willing to discuss some sort of financial compensation for the integration. Really look forward to hearing from you!

carlosperate, 2017-05-22:
Hi @nickedens, you can email me to my personal address and we can discuss further. My email is the same as my github username (carlosperate) at gmail.com

cparrapa, 2017-06-26:
Hi Carlos
 that would be cool i am also interested ;)

BullwinkleII, 2020-03-07:
Is there a way to add _any_ library?

Does Ardublockly use the arduino IDE to upload, and if so, does it mean that I can add the library to the Arduino IDE, and Ardublockly will find it?