carlosperate/ardublockly #172: Building Desktop App on MAC with Pyinstaller

puttley, 2017-05-19:
I am having difficulty building the desktop app on the MAC platform. I can successfully build the entire desktop app on a Windows machine with no issues using py2exe and electron. On the MAC, I can run the build.py script successfully. However, I get error messages when trying to run the build_pyinstaller.py script, and subsequently when using electron npm.

Here is a screen capture of the build_pyinstaller.py shell window:
[image]

I'm hoping to get some assistance on how to re-build the desktop app on the Mac as I've been able to do with Windows.
Thank you.

carlosperate, 2017-05-21:
How are you trying to run the python script? Are you executing the commands from the terminal from the location specified in the instructions?
https://github.com/carlosperate/ardublockly/wiki/Building-Ardublockly#second-step-python-server

I would also highly recommend to use Python3 if possible, but it should still work with Python 2.

puttley, 2017-05-21:
Hi Carlos,
So, just to be clear even though the python script build_pyinstaller.py is located  in the package folder, I should be running the script from the Mac terminal window at the root directory instead?  I run build_py2exe.py on a windows machine from the pacakge folder and all works. I've tried python 2 and 3.

carlosperate, 2017-05-22:
Yes, you'll have to use the terminal at the project root directory instead of double clicking the Python script. This is due to the relative paths present in the PyInstaller spec file inside the package folder.

mariopesch, 2017-05-24:
I am experiencing the same problem. Is there any other fix I could try?

puttley, 2017-05-24:
So, on the MAC I am able to build blockly using build.py script. I am able to run build_pyinstaller.py successfully by running it from the root project directory. And, I am able to build the executable with node. But, when I execute the newly created desktop executable, the MAC desktop Ardublockly window frame opens, but it is blank and just flickers. The log file reports that the server was unable to start and attempts are being made to re-start it. If I replace the server folder in the arduexec folder with the server folder from a build that works, the desktop application will then start. So, even though the pyinstaller.py is executing and reporting successfully build, the server is not being built correctly and/or is unable to start. Here is a screen shot of the log file and also the desktop window that launches blank. Any ideas or help resolving this would be greatly appreciated.
[image]
[image]

noerw, 2017-06-26:
@puttley this seems to be a different issue now.

To summarize my findings about the original issue:
The first error messages are related to `pyinstaller` not beeing installed, fix that by running `pip install pyinstaller` once before running the script. 

When running the script again, pyinstaller complained about the module `ardublockly` not beeing found (both on win10 & ubuntu16 with python 2.7). I could fix that by replacing `import ardublockly` with `import ardublocklyserver` in [`package/pyinstaller.spec`](https://github.com/carlosperate/ardublockly/blob/master/package/pyinstaller.spec#L18).
After that, the build succeeded.

Can anybody reproduce/confirm this bug/fix?