carlosperate/ardublockly #174: 'unicodeescape' Uploading sketch problem

JulesTopart, 2017-06-08:
Hi,
I can't solve this error : 

> Uploading sketch to Arduino...
CLI command: C:\Program Files (x86)\Arduino\arduino_debug.exe C:\\\Users\\jules\\Desktop\\ardublockly-master\\ardublockly-master\ArdublocklySketch\ArdublocklySketch.ino --upload --port COM8 --board arduino:avr:uno
Error: Exception in arduino_ide_send_code:
'unicodeescape' codec can't decode bytes in position 141-142: truncated \UXXXXXXXX escape

This is happenning when I try to upload sketch to my board.
I'm running core version of ardublockly. I study your code a lot an I can't figure out where the problem come from.

- I tryed to run the server with python 3.4 and python 2.7, which doesn't change anything.
- I tryed to change the sketch_path string.
- Execute arduino_debug.exe ... Without anything clear.

Have you any Idea ? 
The problem appeared on a student computer and now it's happening on my own computer too since I totaly format my PC.

Thank's for your help.
Jules

JulesTopart, 2017-06-09:
`----------------------------------------
Exception happened during processing of request from ('127.0.0.1', 49326)
Traceback (most recent call last):
  File "C:\Users\jules\AppData\Local\Programs\Python\Python36-32\lib\socketserver.py", line 317, in _handle_request_noblock
    self.process_request(request, client_address)
  File "C:\Users\jules\AppData\Local\Programs\Python\Python36-32\lib\socketserver.py", line 348, in process_request
    self.finish_request(request, client_address)
  File "C:\Users\jules\AppData\Local\Programs\Python\Python36-32\lib\socketserver.py", line 361, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "C:\Users\jules\AppData\Local\Programs\Python\Python36-32\lib\socketserver.py", line 696, in __init__
    self.handle()
  File "C:\Users\jules\AppData\Local\Programs\Python\Python36-32\lib\http\server.py", line 418, in handle
    self.handle_one_request()
  File "C:\Users\jules\AppData\Local\Programs\Python\Python36-32\lib\http\server.py", line 406, in handle_one_request
    method()
  File "C:\Users\jules\Desktop\server\ardublocklyserver\requesthandler.py", line 62, in do_POST
    message_back = handle_sketch(message_back)
  File "C:\Users\jules\Desktop\server\ardublocklyserver\requesthandler.py", line 184, in handle_sketch
    actions.load_arduino_cli(sketch_path)
  File "C:\Users\jules\Desktop\server\ardublocklyserver\actions.py", line 124, in load_arduino_cli
    error = six.u(error)
  File "C:\Users\jules\Desktop\server\ardublocklyserver\six\six.py", line 599, in u
    return str(s.decode('unicode_escape'))
UnicodeDecodeError: 'unicodeescape' codec can't decode bytes in position 141-142: truncated \UXXXXXXXX escape
----------------------------------------`

New report... same error

JulesTopart, 2017-06-09:
I finally fixed it !
The problem was caused by a string encoding problem. I fix it by modyfying the file six.py line 599
Changing this : 
`return str(s.decode('unicode_escape'))`
into this
`return str(s)`

That's work fine now

JulesTopart, 2017-06-29:
Finally my previous fix was not very accurate, and involve a lot of display problem.
I change the encoding type to 'utf-8' instead of 'unicode-escape' and it work perfectly.

emil01, 2017-07-03:
@carlosperate I faced the same issue when running Ardublockly on python 3.6.1 changing the following lines in actions.py solved the problem for me
**from:**
'
std_out, err_out = process.communicate()
std_out = six.u(std_out)
err_out = six.u(err_out)
exit_code = process.returncode
'
**To:**
`
std_out, err_out = process.communicate()
std_out = std_out.decode("utf-8")
err_out = err_out.decode("utf-8")
exit_code = process.returncode
`

closed 2017-12-26 by JulesTopart