carlosperate/ardublockly #184: Blocks using Fieldnumber with limits are igneored

lazyzero, 2017-07-25:
Hi,

it try to add a block to the set a int value in the range of 0-15. I use the code below, that behaves in the blockfactory of blockly correct. Means that values outside of the limit are set to the max or min limit. It seams that limits are ignored.

Best Christian
```
Blockly.Blocks['matrix_setbrightness'] = {
  init: function() {
    this.appendDummyInput()
        .appendField("Set brightness to")
        .appendField(new Blockly.FieldNumber(15, 0, 15), "BRIGHTNESS");
    this.setPreviousStatement(true, null);
    this.setNextStatement(true, null);
    this.setColour(230);
    this.setTooltip(Blockly.Msg.ARD_SETBRIGHTNESS_TIP);
    this.setHelpUrl('');
  }
};
```

carlosperate, 2017-07-25:
Hi @lazyzero,

Looks like the ability to set minimum and maximum values to FieldNumber was added to Blockly in June 2016 (https://github.com/google/blockly/commit/0be0cc89c7f686ed6377248f7e1caa76578669c3). The blockly version in Ardublockly is a bit older than that, so this feature is not yet supported.

lazyzero, 2017-07-26:
Hi @carlosperate 
thanks for the answer. As your project is injecting own code to the blockly code I guess it is not that easy to update the blockly?

In the meantime I now added a validator that is called onchange events of the block.