carlosperate/ardublockly #187: Ardublockly doesn't remember port

puttley, 2017-08-26:
Hoping for some help on this one. I've found that unlike the Arduino IDE, Ardublockly does not remember the com port selection setting when unplugging/plugging an Arduino UNO. For example, connect UNO to computer and select com port in preferences. If you unplug UNO from computer after downloading code, you then have to re-select the com port every time you re-connect the UNO board. You must also re-select com port under preferences every time Ardublockly is re-started. Is this an issue that can be resolved?

elahroodi, 2018-01-29:
this is my problem too, anyone can help?

carlosperate, 2018-01-31:
This is currently done "by design" (although that doesn't mean it is a good design 😅), as the port is validated every time the application is opened, the settings are opened or a program is uploaded.
However, if you have Ardublockly running, unplug the Arduino do something else, and then replug the Arduino before uploading the program, then it shoudn't have "forgotten" the port.