carlosperate/ardublockly #188: Creating new blocks is not easy

Lokesh-Peri-Shanmugam, 2017-09-06:
creating blocks in google blockly and adding to the code is very tough. Making a tutorial will help developer to bring up many features to ardubockly.

SebCanet, 2017-09-06:
I've tried to explain how to create blocks, but it's in French : [http://info.technologiescollege.fr/wiki/doku.php/fr/arduino/blockly_rduino/creerblocsmultiling](http://info.technologiescollege.fr/wiki/doku.php/fr/arduino/blockly_rduino/creerblocsmultiling)
I started an english version, but it's not at all updated : [http://info.technologiescollege.fr/wiki/doku.php/en/blockly_rduino/create_blocks](http://info.technologiescollege.fr/wiki/doku.php/en/blockly_rduino/create_blocks)
Maybe with a translator it could help you...and maybe you could help Carlos wiki or mine,
Because developpement takes a lot of time I do not have anymore for Blockly documentation, sorry.

Lokesh-Peri-Shanmugam, 2017-09-07:
Thanks SebCanet. I ll try. 👍

wegunterjr, 2017-10-13:
I am working on adding blocks, but I am currently hitting some road blocks.   These steps in your english link are helpful - the ones that show what folders to go in and make changes(I even reference the french one, but I speak Chinese, so....)    but they aren't how I would expect to extend or add a block. They don't show how to compile or run the exe.    Any more details?

SebCanet, 2018-07-07:
You're right, sorry.
In French:
[http://www.libreduc.cc/wiki/doku.php/fr/arduino/blockly_rduino/creerblocsmultiling](http://www.libreduc.cc/wiki/doku.php/fr/arduino/blockly_rduino/creerblocsmultiling)

In english (not updated):
[http://www.libreduc.cc/wiki/doku.php/en/blockly_rduino/create_blocks](http://www.libreduc.cc/wiki/doku.php/en/blockly_rduino/create_blocks)