carlosperate/ardublockly #191: "Error id 52: Unexpected server error." When uploading a Sketch in windows10

RubensQRZ, 2017-09-29:
Dear Carlos, 
a new problem is occurring in the program when we send a code to the Arduio Board using windows 10. Maybe the consequence of some windows update.

The sketch is uploaded, but an error is displayed in the “Arduino IDE Autput”, with the message:

**There has been an error
Unexpected server error.
Error id 52: Unexpected server error.**

Even with this error, you can continue working and sending new sketches do the Arduino Board. But, after each load, the program shows the same error message.

This issue occurs with python 2.7.9 and python 3.4.4. I didn’t try with other versions.  I have made tests in 2 different machines with windows 10-64.

No error message is displayed when we use the “Open Skecth in IDE” option. With this option everything works fine.

The error message displayed in the Python Shell is:

**Uploading sketch to Arduino...
CLI command: C:\Program Files (x86)\Arduino\arduino_debug.exe C:\Users\ruben\Downloads\ardublockly-master\ArdublocklySketch\ArdublocklySketch.ino --upload --port COM3 --board arduino:avr:mega
Error: Exception in arduino_ide_send_code:
'unicodeescape' codec can't decode bytes in position 129-130: truncated \UXXXXXXXX escape**

The complete Python Shell messages  follows below.

Thank you for your attention

best regards

Rubens.
_________________________________________________________

Python 3.4.4 (v3.4.4:737efcadf5a6, Dec 20 2015, 20:20:57) [MSC v.1600 64 bit (AMD64)] on win32
Type "copyright", "credits" or "license()" for more information.
>>> 
======= RESTART: C:\Users\ruben\Downloads\ardublockly-master\start.py =======
Running Python 3.4.4 (64 bit) on Windows-10-10.0.15063
Local packages: C:\Users\ruben\Downloads\ardublockly-master\ardublocklyserver\local-packages

======= Parsing Command line arguments =======
No command line arguments found.

======= Resolving server and project paths =======
Ardublockly root directory:
	C:\Users\ruben\Downloads\ardublockly-master
Current working directory set to:
	C:\Users\ruben\Downloads\ardublockly-master
Selected server root:
	C:\Users\ruben\Downloads\ardublockly-master
Selected server ip:
	localhost
Selected server port:
	8000

======= Loading Settings =======
Settings loaded from:
	C:\Users\ruben\Downloads\ardublockly-master\ServerCompilerSettings.ini
Final settings loaded:
	Compiler directory: C:\Program Files (x86)\Arduino\arduino_debug.exe
	Arduino Board Key: Mega
	Arduino Board Value: arduino:avr:mega
	Serial Port Value: COM3
	Sketch Name: ArdublocklySketch
	Sketch Directory: C:\Users\ruben\Downloads\ardublockly-master
	Load IDE option: upload
Settings file saved to:
	C:\Users\ruben\Downloads\ardublockly-master\ServerCompilerSettings.ini

======= Starting Server =======
Setting HTTP Server Document Root to:
	C:\Users\ruben\Downloads\ardublockly-master
Launch Server:
Bottle v0.12.13 server starting up (using WaitressServer())...
Listening on http://localhost:8000/
Hit Ctrl-C to quit.

Serving on http://DESKTOP-OF0OQIA:8000
Serving on http://DESKTOP-OF0OQIA:8000

Uploading sketch to Arduino...
CLI command: C:\Program Files (x86)\Arduino\arduino_debug.exe C:\Users\ruben\Downloads\ardublockly-master\ArdublocklySketch\ArdublocklySketch.ino --upload --port COM3 --board arduino:avr:mega
Error: Exception in arduino_ide_send_code:
'unicodeescape' codec can't decode bytes in position 129-130: truncated \UXXXXXXXX escape

-----------------------------------------------------------------------------------------------------------

RubensQRZ, 2017-09-29:
After some tests I can see that the problem is with the Arduino Version. This issue does not ocurr if i use Arduino 1.6.11, but with Arduino 1.8.2 it occurs.

puttley, 2018-03-06:
I am having the exact same issue. Can this be resolved? Is there a solution for using the latest Arduino 1.8.5 IDE with Ardublockly on Windows 10?

axeede, 2019-05-13:
Same for me...
Ardunio 1.8.9 with Ardublockly on WIndows 10 Ardublockly v0.1.2

duinopeak, 2019-07-02:
Same for me...

cansik, 2021-03-15:
The problem is the  `six.u` method that does not work as expected. I changed my code to the following and it works now on all OS:

```python
std_out, err_out = process.communicate()

# new conversion code not depending on six
std_out = std_out.decode("utf-8")
err_out = err_out.decode("utf-8")

exit_code = process.returncode
print('Arduino output:\n%s' % std_out)
print('Arduino Error output:\n%s' % err_out)
```
Have a look at the commit I've done in our respository:
https://github.com/IAD-ZHDK/ardublockly/commit/a43797d348044234717b59996ba1d003ec4bf85f