carlosperate/ardublockly #192: Errors when running exe after adding blcoks

wegunterjr, 2017-10-12:
Need some guidance on how to actually do the development for adding blocks. I really want to change how the program is laid out meaning I don't want to have to do a single block for digital or analog. I would really like to just add one that says forward or reverse left or right.. I am doing this for the Sumo bot robot for some 4th and 5th graders B.

wegunterjr, 2017-10-14:
[image]

ongpaulsherng, 2017-12-11:
Hi there! i also meet this problem. anyone who can help?

carlosperate, 2018-01-31:
Is it possible you haven't build the server executable before trying to run the electron application?

Detailed instructions can be found at https://github.com/carlosperate/ardublockly/wiki/Building-Ardublockly

wegunterjr, 2018-02-02:
i followed the steps, and it appears to be building the exe.  When I run it just in the browser, i see the blocks I added, but when I run the exe I generated, it doesn't show the blocks I added. Not sure what I am missing. I followed all the steps except for the packing step (it appeared to only be related to putting that in the cloud).  Suggestions?

carlosperate, 2018-03-11:
Could you check if the `.exe` file is present in the filesystem?
The error message that you previously included shows the path to a `.../arduexec/server/start.exe`. If the PyInstaller step was successfully it should have created that `.exe` file there, could you check that file does indeed exist?