carlosperate/ardublockly #196: Board definitions not being updated on first start

emil01, 2017-12-18:
@carlosperate Upon restarting Ardublockly the I/O pins list is not refreshed. This is particularly noticeable if an ESP based board is selected in preferences. 

Below is the result after restarting Ardublockly once the settings are modified. 

If you could point me in the direction as to where I should be looking I could try and fix it. 

[image]

[image]

carlosperate, 2018-01-31:
Hi @emil01,

Apologies with the late reply. Which version of Ardublockly are you running? As soon as a different board is selected in the setting it should update all the pin options, and if an incompatible pin was selected in any of the blocks a warning message should have been added:

[image]

Can you provide steps to reproduce this? Also, it would really help if you could delete the ardublockly.log file, reproduce the problem, and the copy the file here.

emil01, 2018-01-31:
Hi @carlosperate ,

Once you have set it to ESP8266 WeMos D1, close ardublockly

Now if you could drag and drop the set digital pin block. You will notice that the pins are off.

carlosperate, 2018-01-31:
I see what you mean now, thanks. Yes, the server should inform the front end what board has been selected so that it can update the blocks. I'll need to implement that when I can find a bit of time.

emil01, 2018-01-31:
Thanks, in the meantime could you please point me in the files/folder I should be looking into. I did try to find it myself but was unable to.

gpfernandezflorio, 2020-01-12:
I fixed that requesting the boards in the Ardublockly.init function in ardublockly.js and then calling Ardublockly.changeBlocklyArduinoBoard like in the settings.