carlosperate/ardublockly #200: Server side compilation

fbelleau, 2018-02-20:
Have you planned to offer Ardublocky on a server with compilation server side and a service to download the hex file. Then the only program needed client side would be burning it with avrdude.

This would make Ardublockly very interesting in the class room, no installation  required of Ardublockly and Arduino IDE on each PC. It would be the MicoBit Makecode experience with Arduino or ESP8266.

I have an hacked server now, next step is to download the hex file with appropriate MIME type for avrdude to be run on reception. Would you like to help me modifying the Python server?

http://qliic.blogspot.ca/2018/02/comment-compiler-un-sketch-arduino-avec.html

Tx for this great project.

keptan, 2018-03-01:
@fbelleau 
Hi, I'm interested in your project for education; have you published any more info on your server setup ect.

carlosperate, 2018-03-11:
Hi both,
Thanks a lot for your interest in Ardublockly. The application was never designed to be run on a server, but individually on each user computer (in fact, its original use case was for computers that were offline most of the time).
Because of that, the current design would require a quite a few changes and additions to allow multiple users connect to a single instance of the Ardublockly server. Things like the settings are stored server-side as a single text file, and there is no concept of multiple users or sessions, so all that would have to be redesigned or created.

At the moment I would consider this to be out-of-scope for the project in it's current form, as you still need to download and install something on the users' computers, and also because I am not ready to maintain Ardublockly as a service. But please don't let that discourage you in any way! All the code is open source and everybody is more than welcome to fork it and adapt it to their needs! I would be quite cool to see a project like that, and I'm sure there are people that will find it useful too.