carlosperate/ardublockly #201: Windows 10 and Arduino IDE 1.8.x

puttley, 2018-03-06:
There seems to be an issue with Ardublockly and the 1.8.x versions of the Arduino IDE...specifically 1.8.3 and  higher. When trying to upload a sketch using IDE version 1.8.5, Ardublocky displays error messages in the IDE output window:
**There has been an error
Unexpected server error
Error id 52: Unexpected server error**

When verifying, the Arduino IDE Splash Screen appears, but the verify operation never finishes. The verify button just spins continuosly forever. Is this a known issue? Is there a way to resolve so the latest versions of the IDE can be used?

carlosperate, 2018-03-11:
Hi @puttley, are you sure you are selecting the `arduino_debug.exe` file instead of the `arduino.exe`? The splash screen should not appear with the `arduino_debug.exe`.

puttley, 2018-03-12:
No, I was not selecting arduino_debug.exe. I changed that...thank you,  but the result is still that when clicking verify, ardublockly never completes verification. The verify button spins endlessly. However, you can continue to place blocks while Ardublockly verify/compile button continuous to spin forever.

carlosperate, 2018-03-12:
Could you remove the log file, replicate the issue, and then copy the new log here?
With the `arduino_debug.exe` file it should be able to capture the output from the Arduino executable, that could help me diagnose the issue.

puttley, 2018-03-12:
Carlos,
I've attached the log. Here is a bit more details on what exactly is happening.
When compiling and uploading, the code actually gets uploaded to my target. However, the IDE window is stuck on "waiting for IDE..." and the compile button just continues to spin afterwards continuously. The only way to get the button back working is to select upload sketch to arduino IDE. Once the IDE opens with the sketch, I close the IDE and the compile or verify button can then be clicked again. Same is true for verify. Thank you for your help in getting this resolved.

[ardublockly.log](https://github.com/carlosperate/ardublockly/files/1803412/ardublockly.log)

action-hong, 2018-03-16:
i have some problem,  the IDE is stuck on "waiting for IDE output..."
(Arduino version 1.8.4)

here is my log:

2018-03-16T07:15:05.407Z - info: [ArdublocklyElec] Starting Ardublockly version: 0.1.2
2018-03-16T07:15:05.408Z - info: [ArdublocklyElec] Ardublockly root dir: D:\kk-project\arduino\ardublockly
2018-03-16T07:15:05.408Z - info: [ArdublocklyElec] OS detected: win32
2018-03-16T07:15:05.435Z - info: [ServerMgr] Command: D:\kk-project\arduino\ardublockly\arduexec\server\start.exe --findprojectroot --nobrowser
2018-03-16T07:15:06.083Z - info: [ArdublocklySrv] Running Python 3.4.4 (32 bit) on Windows-10-10.0.14393

======= Parsing Command line arguments =======

Parsed "--findprojectroot" flag. The ardublockly project root will be set as the server root directory.
Parsed "--nobrowser" flag. No browser will be opened.

======= Resolving server and project paths =======

Ardublockly root directory: D:\kk-project\arduino\ardublockly
Selected server root: D:\kk-project\arduino\ardublockly

======= Loading Settings =======

Settings loaded from:
	 D:\kk-project\arduino\ardublockly\ServerCompilerSettings.ini

Final settings loaded:
	Compiler directory: D:\Program Files (x86)\Arduino\arduino_debug.exe
	Arduino Board Key: Uno
	Arduino Board Value: arduino:avr:uno
	Serial Port Value: COM6
	Sketch Name: ArdublocklySketch
	Sketch Directory: D:\kk-project\arduino\ardublockly
	Load IDE option: verify

Settings file saved to:
	D:\kk-project\arduino\ardublockly\ServerCompilerSettings.ini

2018-03-16T07:15:15.206Z - info: [ArdublocklySrv] 

======= Starting Server =======

Setting HTTP Server Document Root to: 
	D:\kk-project\arduino\ardublockly

Launching the HTTP service...

IDE options set to:
	Compile and Upload sketch
Settings file saved to:
	D:\kk-project\arduino\ardublockly\ServerCompilerSettings.ini

elahroodi, 2018-09-03:
Ardublockly works great with new version of Arduino 1.8.6

puttley, 2018-09-04:
The 1.8.6 version upgrade resolved the issue. Thank you.

closed 2018-09-04 by puttley