carlosperate/ardublockly #218: build blockly  fail

easonguolong, 2018-08-30:
Build the first setp fail（build blockly fail）,the log :
SUCCESS: msg\js\tr.js
SUCCESS: msg\js\uk.js
SUCCESS: msg\js\vi.js
SUCCESS: msg\js\zh-hans.js
SUCCESS: msg\js\zh-hant.js
Processed Ardublockly translation: msg\json\en_ardublockly.json
Processed Ardublockly translation: msg\json\es_ardublockly.json
Processed Ardublockly translation: msg\json\fr_ardublockly.json
Processed Ardublockly translation: msg\json\it_ardublockly.json
Processed Ardublockly translation: msg\json\nl_ardublockly.json
Processed Ardublockly translation: msg\json\ru_ardublockly.json
SUCCESS: blockly_uncompressed.js
Exception in thread Thread-2:
Traceback (most recent call last):
  File "D:\python\lib\threading.py", line 530, in __bootstrap_inner
    self.run()
  File "build.py", line 177, in run
    self.gen_core()
  File "build.py", line 210, in gen_core
    self.do_compile(params, target_filename, filenames, [])
  File "build.py", line 274, in do_compile
    conn.request("POST", "/compile", urllib.urlencode(params), headers)
  File "D:\python\lib\httplib.py", line 946, in request
    self._send_request(method, url, body, headers)
  File "D:\python\lib\httplib.py", line 987, in _send_request
    self.endheaders(body)
  File "D:\python\lib\httplib.py", line 940, in endheaders
    self._send_output(message_body)
  File "D:\python\lib\httplib.py", line 803, in _send_output
    self.send(msg)
  File "D:\python\lib\httplib.py", line 755, in send
    self.connect()
  File "D:\python\lib\httplib.py", line 1152, in connect
    self.timeout, self.source_address)
  File "D:\python\lib\socket.py", line 567, in create_connection
    raise error, msg
error: [Errno 10060]

wiener0zyj, 2019-02-22:
the same problem with you!!

ELundby45, 2019-03-05:
Have you guys pulled the latest commit from master? In particular, make sure you have https://github.com/carlosperate/ardublockly/commit/df16522025dafa03d750d88fd24c67979ec3af93