carlosperate/ardublockly #227: Font Issue in Open/Save dialog

Shadouw, 2019-04-19:
Hello,
I'm working on Linux Mint Cinnamon and Kubuntu. Both show the same issue. When I run Ardublockly from shell (ardublockly_run.sh) and click either "Open" or "Save" no letters are visible:
[image]

When I open Ardublockly in Chrome (http://localhost:8000/ardublockly/) it works as expected.

barnycloud, 2019-05-17:
Same problem in ubuntu 18.04 LTS. Ardublockly cannot be used because the compiler location can not be set. Made a new issue in this forum to adress this bug.

Shadouw, 2019-05-17:
@barnycloud , there is at least the workaround with the url in a browser. Besides that you can edit the config file and set the path to the Arduino compiler.

Chacha75, 2019-12-31:
Same problem. Linux Mint 19.3.
Similar problem arised in another electron project:
https://github.com/LightTable/LightTable/issues/2343
and the solution was this libpangoft update:
https://github.com/ArchI3Chris/LightTable-Square-Fix

Chacha75, 2019-12-31:
I've figured out how to do it, and it perfectly works for me.
Clone the 2 files from this github project: https://github.com/ArchI3Chris/LightTable-Square-Fix .
If you download it one by one or in a zip file, then it breaks the symlink, you have to recreate it.
If you have downloaded the **executable** Ardulbockly, copy the 2 files into the `ardublockly/arduexec/ `folder.
If you are using the **source code version**, copy them into the `ardublockly/package/electron/node_modules/electron-prebuilt/dist/` folder.
Happy new year!

Shadouw, 2020-01-03:
@Chacha75 many thanks for this explanation. I can confirm that this fixes my issues on Kubuntu and Mint