carlosperate/ardublockly #233: Error building ardublockly

rbarradas, 2020-02-27:
Hi..
I'm trying to build a customized version of ardublockly and I'm getting this error while executing the second step... python package/build_pyinstaller.py

FileNotFoundError: [WinError 2]
Any ideas on what it could be and how to solve it? 
Thanks

D:\Data\Install\_Programming\Arduino\ArduBlockly\ardublockly-master>"c:\Program Files (x86)\Python38x64\python.exe" package\build_pyinstaller.py
[Ardublockly build] Build procedure started.
[Ardublockly build] Checking for OS.
                    Checking for command line argument indicated OS.
                    Valid command line arg not found, checking system.
                    OS found is: windows
[Ardublockly build] Building Ardublockly for windows.
[Ardublockly build] Project directory is:     D:\Data\Install\_Programming\Arduino\ArduBlockly\ardublockly-master
[Ardublockly build] Script working directory: D:\Data\Install\_Programming\Arduino\ArduBlockly\ardublockly-master
[Ardublockly build] Removing PyInstaller old temp directories.
                    Directory D:\Data\Install\_Programming\Arduino\ArduBlockly\ardublockly-master\dist was not found.
                    Directory D:\Data\Install\_Programming\Arduino\ArduBlockly\ardublockly-master\build was not found.
[Ardublockly build] Running PyInstaller process.
                    Command: ['pyinstaller', 'package\\pyinstaller.spec']
Traceback (most recent call last):
  File "package\build_pyinstaller.py", line 225, in <module>
    build_ardublockly()
  File "package\build_pyinstaller.py", line 196, in build_ardublockly
    success = pyinstaller_build()
  File "package\build_pyinstaller.py", line 104, in pyinstaller_build
    pyinstaller_process = subprocess.Popen(process_args)
  File "c:\Program Files (x86)\Python38x64\lib\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "c:\Program Files (x86)\Python38x64\lib\subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] O sistema não conseguiu localizar o ficheiro especificado

rbarradas, 2020-02-27:
Problem solved...
Basically all had to do with the Path not being set correctly to the Python folders...
Also, after solving this, I realized that one must not install the latest versions of node.js.
Had to install node-v10.0.0-x86.msi and after that, the build process worked fine...
Hope this helps someone with the same problems...

closed 2020-02-27 by rbarradas

chrisaGia, 2023-11-10:
can you say more about : Basically all had to do with the Path not being set correctly to the Python folders.
what we have to change and where?
Thanks!