carlosperate/ardublockly #240: Mac OS Big Sur

puttley, 2021-05-25:
I've recently upgraded my Mac OS to Big Sur which broke Ardublockly. When the app launches all I see is a flashing white box window. Has anyone else had this issue and is there a solution?

RoboThom, 2021-06-27:
[image]
[image]

Although I do not have a solution, only extension.
Maybe someone will be able to better estimate the error from the image I attach.

On the left is version 0.1.2 and on the right is the previous one: 0.1.1.

like puttley, i also have the mac OS Big Sur on the MAC.

the program itself thinks that no functions are implemented and at the end gives a window with the following error message: 

A JavaScript error occurred in the main process

Uncaught Exception:
TypeError: Cannot read property 'webContents' of null
at Timeout._onTimeout (/Applications/Ardublockly.app/Contents/Resources/app.asar/main.js:95:27)
at tryOnTimeout (timers.js:224:11)
at Timer.listOnTimeout (timers.js:198:5)

Regards
Robert

puttley, 2021-06-28:
I have not found a solution, but did come up with a work around. If you start the ardublockly server running first, then launch Ardublockly, the app will work. I modified start.py so that it starts the server, but does not launch the browser version. Once the server is running, start desktop Ardublockly app as normal and it should run. To automate this, I wrote a batch script and added it to my login scripts to run at startup.

bruno0815, 2021-11-14:
Hello @puttley, please excuse me, but could give some more details? I am new in programming and I would like to use ardublocky on my M1, but unfortunately I cannot follow your instructions.

puttley, 2021-11-16:
Sure. Install the app as normal. Navigate to the contents folder and open start.py in an editor. Go to line 77 (I think) and set launch_browser to False. Save start.py.
After you've edited the file, execute start.py. You should see the server starting message in a terminal window. Once the server has started, launch Ardublockly as you normally would.

bruno0815, 2021-11-17:
@puttley Thanks so much. I struggled a little with the server starting part - but finally it works. And it was line 77. 

I did not expect, that you are answering my questions, as this topic is already from June. So thank you very much... Now I can start to learn :)