carlosperate/ardublockly #254: build error

ErNisha4800, 2024-01-16:
Build the first setp 
SUCCESS: msg\js\zh-hant.js
SUCCESS: blockly_uncompressed.js
Traceback (most recent call last):
    self.run()
  File "build.py", line 210, in gen_core
    self.do_compile(params, target_filename, filenames, [])
  File "build.py", line 280, in do_compile
    json_data = json.loads(json_str)
  File "c:\python27\lib\json\__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "c:\python27\lib\json\decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "c:\python27\lib\json\decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded

MrKumakun, 2024-03-11:
Have you run `git submodule update --init --recursive` yet?
https://github.com/carlosperate/ardublockly/wiki/Building-Ardublockly#download-the-source-code

However, some errors still occur, to fix there errors, we had to change the code.
I am committing the fixes to my fork `develop` branch.
https://github.com/MrKumakun/ardublockly-ja/tree/develop/

If it's of any use to you...

raulibanez, 2025-05-20:
@MrKumakun I just can't compile this thing. I even downloaded your dev branch and I get similar errors. I wish I could have a build tutorial with no errors. Could you help?

### python build.py
I just can't get rid of the google closure issue. I tried to run jar, via npx. I just can't run the build.

### Electron build
I get this one. start.exe is missing. I understand this start.exe is coming out of the build.py compilation earlier ?

[18:49:53] Finished 'finalize' after 1.12 ms
[18:49:53] Finished 'bundle' after 22 ms
[18:49:54] Finished 'copy' after 418 ms
[18:49:54] Starting 'build'...
[18:49:54] Finished 'build' after 75 μs
[18:49:54] Starting 'start'...
[18:49:54] Finished 'start' after 257 ms

info: [ArdublocklyElec] Starting Ardublockly version: 0.1.3-a
info: [ArdublocklyElec] Ardublockly root dir: C:\Users\rauli\Dropbox\Repositorios\ardublockly-ja
info: [ArdublocklyElec] OS detected: win32
info: [ServerMgr] Command: C:\Users\rauli\Dropbox\Repositorios\ardublockly-ja\arduexec\server\start.exe --findprojectroot --nobrowser
warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.
warn: [ArdublocklyElec] Page failed to load (-102). The server is probably not yet running. Trying again in 200 ms.

Klein-Kumakiri, 2025-05-21:
@raulibanez

I _think_ the build environment might not be set up correctly (just a guess though...).

The base Blockly version is pretty old, and the build process requires a `Python 2.7` environment.
Also, the server won’t start on Python 3.9 or above (`Python 3.6.8` is recommended).

So, you’ll need to use Python 2.7 when building, and switch to 3.6.8 when running the server.
Try switching between versions using `virtualenv`.

Also, the Electron build environment is kinda tricky — the combination of dependency versions is a bit of a mess.
I had to manually tweak versions of things like `gulp` and `lodash`.

You can check my setup here:
https://github.com/MrKumakun/ardublockly-ja/tree/master/package/electron

MrKumakun, 2025-05-21:
@raulibanez

I think the build environment might not be set up correctly (just a guess though...).

The base Blockly version is pretty old, and the build process requires a Python 2.7 environment.
Also, the server won’t start on Python 3.9 or above (Python 3.6.8 is recommended).

So, you’ll need to use Python 2.7 when building, and switch to 3.6.8 when running the server.
Try switching between versions using virtualenv.

Also, the Electron build environment is kinda tricky — the combination of dependency versions is a bit of a mess.
I had to manually tweak versions of things like gulp and lodash.

You can check my setup here:
https://github.com/MrKumakun/ardublockly-ja/tree/master/package/electron

raulibanez, 2025-05-21:
Thanks a million @MrKumakun 

I had to run python .\package\build_pyinstaller.py with python 3 (I used python 3.9.13) in order to create the missing arduexec/server/start.exe file. I just didn't find that steps in any readme. I've seen it now in the yml fles. Didn't think about that.

Electron is working fine after that.

raulibanez, 2025-05-21:
@MrKumakun I'm curious,

How do you compress the blockly js files?

Build.py is trying to use closure-compiler.appspot.com and it fails compressing because the service is no longer active.

MrKumakun, 2025-05-21:
Got it! @raulibanez 

The steps for `build_pyinstaller.py` are described in the "Second step: Python server" section of the `Building Ardublockly` instructions.
You can find it here, might be helpful:
https://github.com/carlosperate/ardublockly/wiki/Building-Ardublockly

It looks like `closure-compiler.appspot.com` has been shut down.
I couldn’t find any official announcement, but it doesn’t seem to be active anymore.
I hadn’t used it in a while, so I didn’t realize.

Seems like we’ll need to either switch to an offline setup or find another service.
Just in case, here’s a link:
https://mvnrepository.com/artifact/com.google.javascript/closure-compiler